ARMmbed/DAPLink #580: Switch to RTX5

flit, 2019-02-06:
As part of the move to support gcc, we need to switch from RTX4 to RTX5/CMSIS-RTOS2.

Child of #197

eleenest, 2020-06-01:
Hi,

Is there any activity on this task?
We are considering this as an intern project for this summer, if it is not being actively worked on (with the goal of being able to use this on a STM32H based debug board, with other functionality).

Eric

flit, 2020-06-01:
Yes, I have most of the work completed on a branch on my fork: 
https://github.com/flit/DAPLink/tree/feature/gcc

However, it has pretty much stalled since the beginning of the year, mostly due to lack of time. I was actually just today trying to slowly start getting back into it. 😄 

If you would like to talk about intern project ideas related to DAPLink (including this work and gcc), you can email me directly (see [my profile](https://github.com/flit)). There are some really interesting and technically challenging projects that could be defined.

mbrossard, 2023-02-04:
Adding RTX5/CMSIS-RTOS2 was done in #750. Now need to remove old RTX4/CMSIS-RTOS1 (see #1007)

closed 2023-02-04 by mbrossard