ARMmbed/DAPLink #582: Mbed OS DAPLink Port

loverdeg-ep, 2019-02-06:
@flit Feasibility of porting DAPLink to Mbed OS?

- [x] build via mbed-cli

- [ ] Mbed OS RTOS API's? use RTX api's directly?

- [ ] Mbed OS peripheral API compatibility?

- [ ] Mbed OS USB Stack

- [ ] Tests?

flit, 2019-02-14:
Closing this as a duplicate of #197. The individual tasks are linked to that issue.

closed 2019-02-14 by flit

loverdeg-ep, 2019-02-14:
@flit So the issues linked there are all that is needed to port daplink to mbed-os at this point?

loverdeg, 2020-12-10:
@flit Being that Im never satisfied with what actually exists or is remotely close to existing I wanted to capture a question about feasibility of eventually turning this project into an mbed OS library which would contain 'examples' for supported HIC's.

With more MCU's having secure swd support on the horizon it might be interesting to have the capability to integrate daplink into applications or bootloaders. In order to securely update other MCU's on the same PCB.