ARMmbed/DAPLink #585: WinUSB endpoints

Novakov, 2019-02-10:
I'm trying to understand how DAPLink exposes WinUSB-compatible interface. I found how WebUSB with `WINUSB` Microsoft OS Descriptor.  WIndows correctly recognizes it as WCID device and loads WinUSB driver for it.

[image]

According to https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__ConfigUSB__gr.html#WinUSB CMSIS-DAPv2 interface should:
```
CMSIS-DAP with default V2 configuration uses WinUSB and is therefore faster.Optionally support for streaming SWO trace is provided via an additional USB endpoint.The WinUSB configuration requires custom class support with the interface setting – Class Code: 0xFF (Vendor specific), Subclass: 0x00, Protocol code: 0x00. Depending on the configuration it uses the following USB endpoints which should be configured in the interface descriptor in this order:
  Endpoint 1: Bulk Out – used for commands received from host PC.
  Endpoint 2: Bulk In – used for responses send to host PC.
  Endpoint 3: Bulk In (optional) – used for streaming SWO trace (if enabled with SWO_STREAM).
```

USBView however reports no endpoints for WinUSB interface:
```
Interface Descriptor:
bInterfaceNumber:     0x04
bAlternateSetting:    0x00
bNumEndpoints:        0x00
bInterfaceClass:      0xFF
bInterfaceSubClass:   0x03
bInterfaceProtocol:   0x00
iInterface:           0x07
0x0409: "WebUSB: CMSIS-DAP"
```

I code I found: https://github.com/ARMmbed/DAPLink/blob/master/source/usb/usb_lib.c#L1789 which explains why `bNumEndpoints` is zero.

My question: does DAPLink support CMSIS-DAPv2? If so, where can I find bulk endpoints for it?

cederom, 2019-02-10:
Maybe https://github.com/ARMmbed/dapjs would be helpful here? :-)

Novakov, 2019-02-11:
Ok, I think I figured it out. WinUSB support introduced here https://github.com/ARMmbed/DAPLink/pull/409 adds another endpoint that can be used with WebUSB/WinUSB but it's still control transfers,  so no bulk endpoints. So, DAPLink does not support CMSIS-DAP v2 and fast (bulk) WinUSB transfers are not possible.

40Grit, 2019-02-11:
Is CMSIS-DAP V2 and (bulk) WinUSB transfers what makes the FTDI parts viable debug probes?

ArctusGT, 2019-02-14:
I'm in a similar position as @Novakov. In our case, in order to work directly with a MicroPython REPL on the Micro:Bit via WebUSB, we need bulk endpoints to use transferIn/Out instead of controlTransferIn/Out. 

I found an organisation called FiriaLabs have likely forked the DAPLink v243 firmware to enable bulk transfer through WebUSB (https://make.firialabs.com/static/media/maintenance_ab9b0b3.98def0a9.hex - not sure they've shared their changes or have to?) and testing this works as expected.

DAPLink v251:
[image]

FiriaLabs:
[image]

I'm not certain on the specifics needed to make it work, but can we get official support for this in the firmware? Maybe it'll be easier to contribute this after the GCC changes?

Much appreciated 😄

flit, 2019-03-08:
Sorry for the slow reply. The WinUSB support currently in DAPLink is specifically for WebUSB. DAPLink _doesn't yet_ support CMSIS-DAPv2, though that is coming relatively soon. We will also investigate enabling bulk endpoints in addition to control for WebUSB, though we may be running out of endpoints on some of the HICs. Thanks for asking!

thegecko, 2019-05-11:
Added in #629

thegecko, 2019-05-11:
Bulk Endpoint support in DAPjs added in https://github.com/ARMmbed/dapjs/pull/49

loverdeg-ep, 2019-07-12:
@novakov Is this closable now?

@0xc0170

closed 2019-09-23 by 0xc0170