ARMmbed/DAPLink #593: WISE-1570: Firmware unable to reset/halt the target MCU

Fju, 2019-02-25:
I started working with the WISE-1570 MCU and have already succesfully programmed it several times. For drag'n'drop programming I am using the latest DAPLink version (v0253, I also tried v0245) and the WISE-ED22 board for USB connection.
Two days ago the device stopped being programmable. Everytime I drag'n'drop compiled files to the MCU, it fails and shows `FAIL.txt`.
```
The interface firmware FAILED to reset/halt the target MCU
```
I know for sure that the compiled files are correct since I have already successfully programmed them to the MCU!
At first glance, this sounds like a hardware issue since it stopped functioning normally out of nowhere. Nevertheless I hope someone might come up with a solution.

Thanks in advance.
Florian

brianesquilona, 2019-02-25:
@Fju you can try using pyocd to check if CMSIS can connect with your target, and try to flash from there, if not, then must be a hardware issue.

Fju, 2019-02-26:
@brianesquilona thanks for your quick response! I managed to download and install pyOCD, unfortunately it seems like the WISE 1570 board is unknown to pyOCD.
Whenever I want to launch `pyocd gdbserver` it fails and tells me:
```
0000237:WARNING:mbed_board:Unsupported board found 0460
```
What can I do to make it run?

P. S.: I installed pyocd using the setup.py script from the repository (so I guess I should have the latest version)

brianesquilona, 2019-02-26:
You should still be able to read and write the memory using the flash commander.

flit, 2019-02-26:
@Fju Even though pyOCD doesn't know the board, it will still be usable for everything except programming flash.

To investigate the device without needing gdb, you can use the 'pyocd commander' subcommand. Run 'help' to see the list of commands. Of course, you can also use gdb if you prefer.

PS. I'm changing the warning message to be like, "Board ID 0406 is not recognized; you will be able to use pyOCD but not program flash."

Fju, 2019-02-27:
Thanks for the quick response. Unfortunately `pyocd commander` throws an `TransferError` when executed.
```
Exception while initing board: 
Traceback (most recent call last):
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/tools/pyocd.py", line 629, in connect
    self.session.open()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/core/session.py", line 157, in open
    self._board.init()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/board/board.py", line 47, in init
    self.target.init()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 136, in init
    seq.invoke()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/utility/sequencer.py", line 189, in invoke
    resultSequence = call()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 100, in init
    self.read_id_code()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 105, in read_id_code
    self.dpidr = self.read_reg(DP_IDCODE)
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 119, in read_reg
    return self.read_dp(addr, now)
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 234, in read_dp
    return read_dp_cb()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 225, in read_dp_cb
    result = result_cb()
  File "/home/florian/.local/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 254, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/home/florian/.local/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError
```
However pyocd is able to detect the device correctly:
``` sh
$ pyocd list

## => Board Name | Unique ID
-- -- ----------------------                                                                                     
 0 => ARM DAPLink CMSIS-DAP | 0460000015eabbbc00000000000000000000000097969902
```

AriParkkila, 2019-03-22:
@Fju I have noticed that after flashing starts to fail it helps to disconnect/connect WISE-1570 from USB.

When it fails WISE-1570 / FAIL.TXT file has an error message:
```
error: An error occurred during the transfer
type: transient
```

Fju, 2019-03-22:
@AriParkkila Sorry, I don't understand. Do you mean that one should reconnect the device while it's flashing or after a flash has produced the `FAIL.txt` file?

AriParkkila, 2019-03-22:
@Fju for me it helped to disconnect/connect USB **after** WISE-1570 failed to flash and had produced the `FAIL.txt` file. I'm guessing that DAPLink recovery might be due to WISE-1570 is fully powered down, so if you have any other power supply that should probably be removed as well.

Fju, 2019-03-22:
Unfortunately this doesn't change anything. I tried to install different versions of pyocd, but always faced the error that I have already posted when executing `pyocd commander`.