ARMmbed/DAPLink #594: STM32F103xb HIC prompt not recognize

Fairchi1d, 2019-02-27:
Dear All,
I made a daplink using stm32f103xb scheme. During debugging, I found a strange phenomenon: if I directly connect the device to the PC, it will prompt cannot recognize.  But power supplied to device first, then connected to PC successfully identify.  Do you have similar problems?  I compiled boot and app on REL v0251 version.

 I still have a another question, why STM32F103xB has been supported after version v251, but the official release still does not contain the corresponding bin file.Is the daplink project based on stm32 immature?  Not fully verified?

Best Regards
Charon

brianesquilona, 2019-03-04:
I'm not really sure why you are observing this, seems like a hardware issue?.

As for your other question, I have tested a STM32F103xB  on a STM32F103rB  target so I can say that it is stable.

Fairchi1d, 2019-03-05:
I have fixed the problem and modified it as follows: 
[bootloader]
#define USB_CONNECT_DELAY       (0)    //11->0

[interface]
#define USB_CONNECT_DELAY       (0)    //11->0
//busy_wait ()   //gpio_init() interface remove this line，but bootloader still keep

I did this just to reduce the delay time of running usbd_connect(1), I'm not sure this change whether correct?

closed 2021-05-27 by mathias-arm