ARMmbed/DAPLink #595: Which STM MCU does DAPLINK support?

heejungPark, 2019-02-27:
I just wonder that DAPLINK supports STM MCUs.
If supports, which MCU is supported?

Thanks.

brianesquilona, 2019-02-27:
@heejungPark You can check [projects.yaml](https://github.com/ARMmbed/DAPLink/blob/master/projects.yaml), stm32f103xb is supported for hic and targets are in [st family](https://github.com/ARMmbed/DAPLink/tree/master/source/family/st).

Fairchi1d, 2019-02-28:
> @heejungPark You can check [projects.yaml](https://github.com/ARMmbed/DAPLink/blob/master/projects.yaml), stm32f103xb is supported for hic and targets are in [st family](https://github.com/ARMmbed/DAPLink/tree/master/source/family/st).

Hello, I have met some problems, could you help me? The issues submitted are #594, thank you

bagyavinoth, 2019-04-16:
> @heejungPark You can check [projects.yaml](https://github.com/ARMmbed/DAPLink/blob/master/projects.yaml), stm32f103xb is supported for hic and targets are in [st family](https://github.com/ARMmbed/DAPLink/tree/master/source/family/st).

@brianesquilona :
I use this evaluation board : [Link](https://www.st.com/en/evaluation-tools/stm32f779i-eval.html)
Where the debug unit MCU is "STM32F103CBT6" & the application MCU is "STM32F779NI". 
Debug unit supports ST-LINK firmware which is a closed source. I would like to use "DAPLink" in the debug unit.
As per my understanding, HIC stm32f103xb should support. Right ? Ofcourse the target support to be added for "STM32F779NI"

But my development environment is GCC_ARM. Can i build DAPLink with GCC_ARM ?

closed 2021-05-27 by mathias-arm