ARMmbed/DAPLink #597: Problem when IR length is 8 bit

PetShelford, 2019-03-05:
I've been trying to set IR length 8 bit, but it leads to invalid scans of IDCODE, APACC, DPACC and ABORT registers. The problem actually is in 'C- defines' where 4 MSB bits defined as '0'. The chip I'm targeting wants it as '1'. So the question: does anybody try to use DAPLink with target which has 8-bit IR length?

flit, 2019-03-05:
What DAPLink build are you using that supports JTAG?

PetShelford, 2019-03-06:
Well, I've been told it is CMSIS-DAP v5_224. It is for LPC-Link2 and is a part of LPCScrypt software.

flit, 2019-03-07:
I *think* that is the NXP-proprietary firmware for LPC-LinkII. DAPLink is different, open source, firmware developed by Arm. I know the NXP tools team, so let me follow up on it and I'll get back to you here.

flit, 2019-06-13:
What device are you using that has an 8-bit IR? We've never seen a device with this IR length yet.

Even though the firmware you are using is not DAPLink, we want to ensure that >4-bit IR length is supported in DAPLink (and other Arm debug probes like ULINK).

And sorry, I never heard back from the NXP tools team! However, the LPC-LinkII firmware is known to be using the reference CMSIS-DAP code released as part of the CMSIS package. So if an 8-bit IR doesn't work with LPC-LinkII, then it probably won't work with any other probe that uses the reference CMSIS-DAP code.

flit, 2019-06-20:
We've confirmed that it's the reference CMSIS-DAP code that does not support an 8-bit IR.

closed 2021-06-17 by mathias-arm