ARMmbed/DAPLink #602: Question about Compatibility with MCUs

joelvonrotz, 2019-03-08:
Hi ARMmbed

I'm currently working on a LPC1768 devboard (custom PCB) and plan on adding DAPLink to the second board. I had a peek at the SWDAP-LPC11U35 and DIPDAP-LPC11U35 boards and now looked through the DAPLink Repository. Are the two boards still up-to-date?

With which MCUs are they compatible? The Handbook states four MCUs (NXP K64F, NXP LPC1768, Nordic nRF51822 and Multitech xDOT), are there any other MCUs compatible?

flit, 2019-03-08:
DAPLink firmware supports any target MCU through the CMSIS-DAP interface. The host provides all the target support; CMSIS-DAP is just a conduit for SWD/JTAG.

For the MSD drag-and-drop flash programming interface, DAPLink supports a large number of targets. It's easiest to just point you at [the source directory](https://github.com/ARMmbed/DAPLink/tree/master/source/family). We are also very quickly moving to where you will be able to run a script to bind a flash algo from a CMSIS-Pack into a prebuilt binary in order to support any target with a Pack. If you don't need the MSD interface, you can use the "no-target" build for your HIC.

chris-styles, 2019-03-08:
For the hardware, you should be looking at 

https://github.com/ARMmbed/mbed-HDK-Eagle-Projects/tree/master/DAPLink

There are two subcategories... Interface schematics which are logical designs, and "probes" which are physical implementations.

As you are doing your own board design, I would suggest you'd want to start with the latest LPC11U35 logical design (v1.1.0) and integrate that into your project.

The Probes (such as SWDAP, DAPule), are just full designs we've made, mostly for our own internal use, but published them as they might be useful.

joelvonrotz, 2019-03-09:
@flit 
Thank you very much! As long as the Target has the required pins they can be used with DAPLink, is that right?

@chris-styles Sounds promising, thanks for the response!

closed 2019-03-09 by joelvonrotz

flit, 2019-03-09:
That's right, DAPLink will support any target with the required pins. Those are SWDIO and SWCLK, which should be available on any Cortex-M MCU. A hardware reset pin (nRESET) is also nice to have, but not required.

joelvonrotz, 2019-03-09:
> 
> 
> That's right, DAPLink will support any target with the required pins. Those are SWDIO and SWCLK, which should be available on any Cortex-M MCU. A hardware reset pin (nRESET) is also nice to have, but not required.

Nice, the more the better. Thank you so much for taking the time to help me ☺️ 

Have a nice day!