ARMmbed/DAPLink #606: the custom DAP Vendor Command IDs range

wjzhang, 2019-03-13:
Hi:
     when we want add some custom DAP Vendor Commands, because of DAPLink also used ID_DAP_Vendor0/.../ID_DAP_Vendor31.  
    is ID_DAP_Vendor0/.../ID_DAP_Vendor31 reserve for ARM DAPLink official use?
    what's about using ID_DAP_Vendor31 down count? 
    what's IDs range safe? from 0xA0 -- 0xFE?

flit, 2019-03-13:
Currently we have not thought about supporting target-specific CMSIS-DAP vendor commands. Also, there is not currently a method for target-specific code to add vendor commands. This would be required so that the common `DAP_vendor.c` file doesn't have to be modified. (That's pretty easy, just a weak function called in the `DAP_ProcessVendorCommand()` default switch case, it just needs to be done.)

Can you provide some context about what target and the commands you want to add?

wjzhang, 2019-03-14:
we want add some only used by ourself vendor commands, such as get full uniqueID, change LEDs....
and wanted avoid conflict with DAPlink add office vendor commands in the future.
look for safe vendor commands IDs range.

flit, 2019-03-19:
I think you'll be safe if you use commands 0xA0 through 0xFE, but start at 0xFE and count down.

closed 2019-03-25 by wjzhang