ARMmbed/DAPLink #609: 1.3 micro"bit fails to enter automation mode

microbit-mark, 2019-03-19:
With automation enabled, flashing an empty `start_bl.act` file should reboot the micro:bit into MAINTENANCE mode. On 1.5 board variants this works, but on 1.3b board variants, it fails.

 **To reproduce on 1.3b**
 - Confirm DETAILS.txt file contains “Automation allowed: 0”
 - Copy an empty `auto_on.cfg` onto micro:bit while reset button is pressed
 - Release reset button while usb drive remounts
 - Confirm DETAILS.txt file contains “Automation allowed: 1”
 - Copy an empty `start_bl.act` file to `MICROBIT` drive and it should reboot into `MAINTENANCE`
On 1.3b variants it reboots into normal MICROBIT mode.

**Expected result**
All micro:bit variants should facilitate automation using steps listed above.

**Environment**
MacOS 10.14.3
The 1.5 board has Bootloader v0246 and the 1.3b in maintenance has v0234

**Additional context**
- 1.3b exhibits the issue with DAPLink 0241 through 0253.  I have also tried this on another 1.3B.
- The manual process of holding reset and adding the `start_bl.act` file does enable MAINTENANCE mode.

brianesquilona, 2019-03-19:
@microbit-mark, can you send us board/s for this one. Feel free to email me, because I might need more information about these boards.

microbit-pauline, 2019-03-21:
@brianesquilona I have shipped 3 micro:bit boards for your attention.

brianesquilona, 2019-04-01:
@microbit-mark bootloader 0234 old and does not support this switching, you can update your bootloader to 0250 or 0251. Depending on your interface code, after 0250, the memory map of the kl26z for bootloader/if changed to accomodate added logics. But one of these should work.

[kl26z_bl.zip](https://github.com/ARMmbed/DAPLink/files/3030890/kl26z_bl.zip)

microbit-carlos, 2019-04-08:
Just to clarify, with a micro:bit 1.3B DAPLink bootloader 0234 and interface 0241, copying the `start_bl.act` to the interface drive while the reset button is pressed does work, and DAPLink reboots into bootloader mode.

But if automation is enabled and the same `start_bl.act` file is copied to the interface drive without pressing the reset button, it does not reset into bootloader mode. It doesn't really do anything, just ignores the file.

Wouldn't the logic that decides when to act on this file be present in the interface? Not sure why an older version of the bootloader would affect this, if it works when the reset button is pressed.

> Depending on your interface code, after 0250, the memory map of the kl26z for bootloader/if changed to accomodate added logics. But one of these should work.

Does that mean that versions of the bootloader < 0250 wouldn't work with version of DAPLink > 0251? Or is it just some features that are affected?