ARMmbed/DAPLink #610: Firmware doesn't fit in FRDM-K66F

zepperik, 2019-03-19:
When mounted as a "bootloader" drive, Windows claims that it needs 4kB of extra space.  I need to boot into Linux to successfully copy the firmware.

brianesquilona, 2019-03-19:
@zepperik, if you have flash the interface, try to drag and drop the k20dx_bl bootloader to update it. New bootloader drive will have "MAINTENANCE" drive name. Let me know if you need the binary.

zepperik, 2019-03-19:
The link: https://armmbed.github.io/DAPLink/?board=FRDM-K66F gives me a file called 0253_k20dx_frdmk66f_0x8000.bin.  (This link also included this github link.)  The board shows up as BOOTLOADER when plugged in with the reset button held down.  Dragging and dropping this file into the BOOTLOADER drive gives me the error about not having enough space.

brianesquilona, 2019-03-19:
[k20dx_bl_crc.zip](https://github.com/ARMmbed/DAPLink/files/2984373/k20dx_bl_crc.zip)
You can try this bootloader to update the bootloader in K66F, after the interface update of 0253_k20dx_frdmk66f_0x8000.bin. Flash this bin in interface mode. Let me know if you were able to change the bootloader (to MAINTENANCE) by this method.

zepperik, 2019-03-19:
Thanks.  I already updated it by copying the binary while in Linux.  I didn't expect such a quick response.  I was going to update the bootloader, but in the link: https://os.mbed.com/blog/entry/DAPLink-bootloader-update/ the FRDM-K66F is not listed so I didn't want to risk bricking another board.

brianesquilona, 2019-03-19:
[k20dx_bl_crc.zip](https://github.com/ARMmbed/DAPLink/files/2984592/k20dx_bl_crc.zip)
@zepperik 
I think I have made a mistake, the master branch have a different memory map now for the bootloader so the last one will not work, try this one instead.

closed 2021-06-08 by mathias-arm