ARMmbed/DAPLink #611: CDC break reset doesn't work with SW reset

flit, 2019-03-19:
A CDC break to trigger target reset does not properly reset targets that use software reset. The underlying issue is that `swd_set_target_reset()` in `target_family.c` does not support software reset. The call sequence goes `target_set_state(RESET_HOLD)` –> `target_set_state_sw(RESET_HOLD)` –> `swd_set_target_reset()`, which then simply controls the nRESET pin.

Part of the problem here is that `SYSRESETREQ` and `VECTRESET` are both auto-clear bits, so you can't hold the core in reset using these controls.

Tasks:
1. The logic to support software reset will have to be moved higher.
2. Make `swd_set_target_reset()` handle software reset? Maybe it should just cause an immediate sw reset. Or perhaps it should just assert.
3. `target_set_state_sw()` must assert if called with `RESET_HOLD` since that is not valid.

cederom, 2019-03-20:
Hey @flit have you seen https://github.com/ARMmbed/DAPLink/pull/371 ? Maybe you will find anything handy there to cherry pick ;-)

flit, 2019-03-20:
@cederom Thanks for the reminder of your PR!