ARMmbed/DAPLink #621: DAPLink BootLoader v 243 will not allow JLink to be installed on a Micro:Bit

leematos, 2019-03-31:
Hi!

I'm trying to get JLink on a Micro:Bit. It looks like I'm running:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0000000050494e4500622003000000160000000097969901
HIC ID: 97969901
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0243
Interface Version: 0253
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0x53375800
Remount count: 0
```

When I try and flash JLink onto the device I get the following failures:

FAIL.TXT
```
An internal error has occurred
```

ASSERT.TXT
```
Assert
File: (file)
Line: 206
Source: Bootloader
```

Post-JLink Failed Flash DETAILS.TXT
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0000000050494e4500622003000000160000000097969901
HIC ID: 97969901
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0xd87a6179
Remount count: 0
```

Based on the error it seems like something with Bootloader v243.

I asked a friend who I know had an older Micro:Bit. He confirmed his bootloader version:
```
Version: 0234
Build:   Oct 12 2015 14:39:34
```

We Flashed JLink and he got this message, which I assume means it worked:

```
This J-Link VMSD config can be used to flash a NordicSemi nRF51822 target device.
```

I cannot find any resources on:

1. Debugging this Bootloader error any further. (I'm also an embedded newb)
1. If it's possible to revert Bootloader versions in DAPLink.

If anyone could help with either of those, I'd really appreciate it. It's that _really close_ feeling! 💃

leematos, 2019-03-31:
Was just inspired, and maybe it's:

https://github.com/ARMmbed/DAPLink/blob/0243/source/daplink/bootloader/main.c#L206

Also for reference, I've tried flashing on a Mac and a PC, so I don't think this is a problem with a specific USB port, also tried 2 different cables. <3

microbit-carlos, 2019-04-01:
Link to the hex file: https://www.segger.com/downloads/jlink#BBC_microbit

The j-link hex file does work with DAPLink 0234 bootloader, but not with the newer 0243.

One simple difference in this hex file is that it has 32 data bytes per record instead of 16, but that's not the problem, as converting a DAPLink hex file to the same format using srec_cat produces a hex file that DAPLink 0243 bootloader can flash correctly: [0250_kl26z_microbit_0x8000-hex32.hex.zip](https://github.com/ARMmbed/DAPLink/files/3029712/0250_kl26z_microbit_0x8000-hex32.hex.zip)

And converting the j-link file into a 16 byte records format produces a file that still cannot be flashed. (Not sharing the output file in case it goes against the J-link license agreement)

Converting the j-link hex file into a bin file works, but that's less surprising and it's just a dump of data to flash.

leematos, 2019-04-01:
@microbit-carlos thanks for confirming I'm not insane! 💃 

> Converting the j-link hex file into a bin file works, but that's less surprising and it's just a dump of data to flash.

1. I assume this means "the data will flash, but the program doesn't work"?
2. If that assumption is wrong, I can use that as a stop-gap, as we work to figure out what the root of the problem is?

I see http://www.keil.com/download/docs/7.asp to convert intel hex to bin but I'm not sure the right flags needed. 

As a newb if this "solves my problem" and you have steps to do so, I'll happily get out of your hair 😸

microbit-carlos, 2019-04-01:
The bin file worked, but I tried this a while ago and I don't remember which tool or flags I used. It's quite possible I used the hex2bin tool from the the Intel Hex Python library, as I have that handy and use it often (https://python-intelhex.readthedocs.io/en/latest/part3-1.html). You shouldn't need anything fancy, just a bin file that starts at address 0.

leematos, 2019-04-01:
Thanks! I'll ask 1 more question to fill in my knowledge gaps:

Based on the above, it seems like I do not need to specify a "length" for this bin file. If I would, of hand, do you know what that length would be for the microbit? I'm ass/u/me/ing it would be standard , but can't hurt to ask.

leematos, 2019-04-01:
For posterity, for anyone else who might run into this:

literally:

python hex2bin.py <jlink.hex> jlink.bin

It flashed! Now to actually try and use JLink 👹

0xc0170, 2019-11-08:
Can this be closed or is there any action outstanding?

microbit-carlos, 2019-11-08:
I haven't tested it with the latest DAPLink bootloader, but as far as I am aware the problem hasn't been solved, just a workaround found (to convert the original J-Link hex file into a bin file).
Used to work with bootloader 0234, but anymore after 0243.

The hex file causing issues can be found here: https://www.segger.com/downloads/jlink#BBC_microbit