ARMmbed/DAPLink #626: DAPLink based on STM32F103xb can't reset after programming

LagrangeWang, 2019-04-30:
I have a DAPLink based on STM32F103CBT6, when I program into another chip such as STM32F103C8T6 with the keil5 and I have checked "reset and run",after programming the STM32F103C8T6 cannot auto reset and run.

toyowata, 2020-05-01:
@LagrangeWang  You can enable auto reset mode.

https://github.com/ARMmbed/DAPLink/blob/master/docs/MSD_COMMANDS.md#configuration-commands

XueYuSky, 2020-06-21:
> @LagrangeWang You can enable auto reset mode.
> 
> https://github.com/ARMmbed/DAPLink/blob/master/docs/MSD_COMMANDS.md#configuration-commands

I wander that  wether reset pin connected is essential or not?   If I config auto reset  to 1,in the DETAILS.TXT， and connect the  reset Pin of dap-link and target board , it works . If I do not connet reset Pin , it dose not work.   But  jlink -swd   do not neet that.
is that right ?

RadioOperator, 2020-06-21:
try this one:
https://github.com/bh3nvn/DAPLink253
or this:
https://github.com/RadioOperator/STM32F103C8T6_CMSIS-DAP_SWO