ARMmbed/DAPLink #633: Windows does not recognize the device

aklinz, 2019-05-19:
Hi
Use MDK v5.26. STM32F103R8
When I opened the generated MDK project for the first time, "This is an version 4project..." popped up. After clicking "Migrate to Device Pack", there is no problem.
I compiled the MDK project successfully.Project download via JLink.
However, after connecting via USB, it prompts
"Windows has stopped it because of a problem with the device. (Code 43)
Requesting a USB device descriptor failed. "

Have tried changing the cable. Or start the MCU again connect to the computer.Have failed.

aklinz, 2019-05-19:
Generated a lot of folders like "stm32f103xb_stm32f103rb_if".
I opened "stm32f103xb_stm32f103rb_if" project and simply changed the Device option from f103rb to f103r8.

0xc0170, 2019-10-21:
@Aakrin Have you resolved this issue?

Unfortunately, I dont have enough details to help here.

closed 2022-03-17 by yfclks