ARMmbed/DAPLink #634: How to run on other MCUs, such as stm32f401cc

aklina, 2019-05-21:
After reading the doc, I still don't know how to transplant.
I currently only know that you may need to change flash_blob.c, target.c, and there seems to be a .yaml file. Where to change, how to change, the method is very vague. If I follow the documentation, it is absolutely impossible to succeed.
Is there a complete process or step to guide me through the modification of the project?
If so, I think it can arouse more people's interest, including at least me.

0xc0170, 2019-05-21:
@aklina Can you point out more specific with urls and details what is not clear? This could help us to revise what we have and should have.

aklina, 2019-05-22:
@0xc0170 Hi
In docs/PORT_TARGET_FAMILY.md, you need to change flash_blob.c and target.c for different targets,
and a [barod].c file for the target in DAPLink\source\board, for example stm32f401re.c.
“A flash algorithm blob is needed to program the target MCUs internal (or external)”,This algorithm does not know where to get it.

Except for these files, it seems that other files are the same.
I want to make DAPlink run on stm32f401cc, how can I modify the file?
I want to know the specific content of the modification, because different chips need to match the program.

mathias-arm, 2021-06-08:
Sadly, [docs/PORT_HIC.md](https://github.com/ARMmbed/DAPLink/blob/master/docs/PORT_HIC.md) remains so far empty.

Recently support for new MCUs have been added. This [branch](https://github.com/mbrossard/DAPLink/tree/feature/stm32f072cb_hic) is a work in progress to add support for STM32F072CB.

I was thinking of trying to port to STM32F401CE as an experiment and I know the peripherals of the family are very similar to STM32F103. So the process would probably be to take `source/hic_hal/stm32/stm32f103xb` as an inspiration and replace the  startup/system/HAL files with the correct ones from ST, and modify the rest to fit the new target. One thing that was different was Flash where STM32F4 has large sectors with different sizes.

closed 2021-06-08 by mathias-arm