ARMmbed/DAPLink #637: When I compiled, I found a bug in post_buind_script.py

cshsoft, 2019-05-29:
Program Size: Code=28394 RO-data=4118 RW-data=848 ZI-data=9612  
After Build - User command #1: ..\..\..\tools\post_build_script.bat G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl C:\Keil_v5\ARM
Python 3.7.2
G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl>fromelf --bin G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl.axf -o G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl.bin 
G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl>fromelf --i32 G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl.axf -o G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl.hex 
G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl>python ..\..\..\tools\post_build_script.py G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl.hex G:\Projects\2019\DAPLINK_DEV\DapLink\DAPLink_csh\projectfiles\uvision\stm32f103xb_bl\build\stm32f103xb_bl_crc 
  **File "..\..\..\tools\post_build_script.py", line 144
    pack_flash_algo.symbols['EraseChip'] + blob_header_size + flash_blob_entry if pack_flash_algo.symbols['EraseChip'] != 0xffffffffL else 0,**
                                                                                                                                    ^
SyntaxError: invalid syntax
".\build\stm32f103xb_bl.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:08

cshsoft, 2019-05-30:
I fixed this bug..
Because of Python 3, we need to remove the L after 0xffffffffffl, and then modify StringIO in flash_alogo.py to be IO

cshsoft, 2019-05-30:
Enjoy

0xc0170, 2019-06-04:
@cshsoft Thanks for providing also the fix. Can you please send PR addressing this issue?

mathias-arm, 2021-06-08:
Fixed in [19a5f33] by @flit

closed 2021-06-08 by mathias-arm