ARMmbed/DAPLink #640: Failed to program target: Assert in target_flash_erase_sector_size - stm32f103xb_stm32l476rg_if

efopee, 2019-06-07:
Can anyone confirm that the stm32f103xb_stm32l476rg_if project works properly, using STM32F103CBT6 as HIC, and an STM32L476RG as target?

I tried using a DAPlink interface on a NUCLEO-L476RG board. Both the HIC and the target MCUs are supported in the codebase, a corresponding project already exists in the repo, as written above.

I compiled the project and downloaded it to the interface MCU on the Nucleo board using an ST-LINK programmer. After this, the DAPLINK USB drive popped up on my Windows machine. I tried programming the target MCU using the drag-and-drop method, and failed.

An assert is raised and documented in ASSERT.txt, which points to line 378 in the function `target_flash_erase_sector_size()` in [](https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/interface/target_flash.c)

Has anyone used or tested this combination with DAPlink before, thereby confirming, that the target support is correct? Where should I start digging to find a solution to this?

psandelin, 2019-06-19:
I had the same assert when doing Drag-n-Drop programming a STM32 target.
Try the following workaround in 
[https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/flash_decoder.c](daplink/drag-n-drop/flash_decoder.c)
Line: 149
Function: flash_decoder_get_flash()
`#if 0 // bug in Daplink? `
`                    if (addr >= region->start &&  addr <= region->end) {`
`#else //  work around when addr is 0.`
`                    if (/*addr >= 0 &&*/  addr <= (region->end - region->start)) {`
`#endif`

They seemingly broke something in Drag-n-Drop when refactoring a few months ago.

0xc0170, 2019-10-21:
@psandelin Thanks for providing the workaround.

@efopee Did the above suggestion fix your problem ?

0xc0170, 2019-10-21:
This was fixed on master (line 148 now in flash_decoder.c file)

psandelin, 2019-10-21:
@efopee, yes, I saw and tried the current fix on master and it does indeed fix the issue I saw. Thank you.

closed 2019-10-21 by 0xc0170