ARMmbed/DAPLink #644: Arm Compiler 6 Support AC6

loverdeg-ep, 2019-07-12:
@ARMmbed/team-embeddedplanet  cannot build daplink since we do not have AC5.
I am expressing our desire to at least be able to build with AC6.

40Grit, 2019-08-10:
If someone could get this ported to at the very least ac6 @team-embeddedPlanet could start making contributions to this project.

@maclobdell @flit @blackstoneengineering

0xc0170, 2019-08-12:
How many errors are there (I haven't yet tested it with AC6) ?

I can help fixing some but can't commit to fully enable this at the moment 😢

40Grit, 2019-08-12:
We tested Ac6 once and it seemed non-trivial to port,  However we were in a rush at the time.

We have initial protos coming in for a board that we want to make into a product for generic debug and device flashing.
It will also provide an interface for our Mbed Enabled + Pelion ready modules and kits.

Unless we are missing something simple that will enable ac6 to build daplink, we wont be able to use our protos when they come in this week.

@brianesquilona

0xc0170, 2019-08-12:
My first thought was about our current RTX here (RTX4) being able to support AC6. We rather move to RTX5. I'll catch up with @brianesquilona if we can switch (there is size increase as I was told so might not be able to do it completely).

loverdeg-ep, 2019-08-12:
@millerc-ep

brianesquilona, 2019-08-16:
Yes, switching to RTX5 will have size increase but bootloaders are already in the no-rtos codes, so only the interface codes will need to be migrated. All the interfaces/headers conforms to RTX5 already.

millerc-ep, 2019-08-20:
We tried building daplink with mdk5-lite but the build is too big for it.
[image]

0xc0170, 2019-08-21:
The current master builds with lite (32k linker limit) for your interface?

millerc-ep, 2019-08-22:
Correct, we are trying to build it but we are hitting the 32k linker limit.

loverdeg-ep, 2019-08-22:
@0xc0170  We have not swapped RTX4 for RTX5 nor tried. 

We want to be able to contribute back to the project in it's current state so we can add support for our Debug & Test board sooner than later.

The swap would imply getting our changes onto master and getting that released which I presume will take a while.

mathias-arm, 2020-12-15:
I have been working on this in [https://github.com/mathias-arm/DAPLink/tree/feature/armclang](feature/armclang). It builds on the GCC support @flit started. To test you need also the changes in project-generator/project_generator#474.

mbrossard, 2020-12-17:
Hi everyone, I just wanted to draw your attention to #750.

Note: I have only tested with the new `make_armclang` toolchain option from project-generator/project_generator#475, there is also a `cmake_armclang` option with project-generator/project_generator#475)

Once you have installed the Python requirements and the proper version of `project-generator`:
```
$ progen generate -t make_armclang -p <project-name>
$ make -C projectfiles/make_armclang/<project-name>
```

closed 2023-02-04 by mbrossard