ARMmbed/DAPLink #645: Keil MDK output fatal: Needed a single revision

Aakrin, 2019-08-02:
```
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'D:\Data\Computer\Embedded\Stm32\Tool\Keil\ARM\ARMCC\Bin'
Build target 'stm32f103xb_stm32f103rb_if'
Before Build - User command #1: ..\..\..\tools\pre_build_script.bat
C:\Users\User\Desktop\DAPLink-master\projectfiles\uvision\stm32f103xb_stm32f103rb_if>python ..\..\..\tools\pre_build_script.py 
fatal: Needed a single revision
#> Pre-build script start
#> Getting git SHA
#> ERROR: Failed to get git SHA, do you have git.exe in your PATH environment variable?
".\build\stm32f103xb_stm32f103rb_if.axf" - 1 Error(s), 0 Warning(s).
Target not created.
```
```
PS C:\Users\User\Desktop\DAPLink-master> git
usage: git [--version] [--help] [-C <path>] [-c <name>=<value>]
           [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p | --paginate | -P | --no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           <command> [<args>]

These are common Git commands used in various situations:

start a working area (see also: git help tutorial)
   clone      Clone a repository into a new directory
...Omit a lot.
```
How to solve this problem?

0xc0170, 2019-08-12:
These lines are failing: https://github.com/ARMmbed/DAPLink/blob/master/tools/pre_build_script.py#L62

Can you run it again with some checks for lines 65 and 66 - what exceptions are you getting?

Aakrin, 2019-08-14:
@0xc0170 
This is the result of running in vscode
```
[Running] python -u "c:\Users\User\Desktop\DAPLink-master\tools\pre_build_script.py"
#> Pre-build script start
#> Getting git SHA
#> Checking for local changes
#> Writing git version file
Traceback (most recent call last):
  File "c:\Users\User\Desktop\DAPLink-master\tools\pre_build_script.py", line 103, in <module>
    exit(generate_version_file(args.version_git_dir))
  File "c:\Users\User\Desktop\DAPLink-master\tools\pre_build_script.py", line 90, in generate_version_file
    with open(output_file, 'w+') as version_file:
IOError: [Errno 2] No such file or directory: '..\\..\\..\\source\\daplink\\version_git.h'

[Done] exited with code=1 in 10.405 seconds
```

0xc0170, 2019-10-21:
@Aakrin Any update?

Aakrin, 2019-10-22:
The solution is to disable the python file from running in MDK

0xc0170, 2019-10-22:
Thanks for sharing. Considering this as resolved.

closed 2019-10-22 by 0xc0170

e0r, 2020-09-21:
I have found this question too

e0r, 2020-09-21:
> I have found this question too

just run this 2line in pre_build_script.bat by your hand:
python %TOOLS%\pre_build_script.py
copy %DIR%\version_git_tmpl.txt %DIR%\version_git.h
and then delete it.
have not found better methond.

Looomo, 2021-08-11:
Use git clone to download this project, instead of downloading the ZIP file.
For me, this error was due to the execution failure of "git rev-parse --verify HEAD" in pre_build_script.py, line 68, which needs the .git folder.

LONG300002, 2023-10-07:
> "D:\keil5 mdk\core\UV4\UV4.exe" -b f:\1\DAPLink-main\projectfiles\uvision\stm32f103xb_if\stm32f103xb_if.uvprojx -j0 -t stm32f103xb_if -o f:\1\DAPLink-main\projectfiles\uvision\stm32f103xb_if\.vscode\uv4.log

*** Using Compiler 'V5.06 update 7 (build 960)', folder: 'D:\keil5 mdk\core\ARM\ARMCC\Bin'
Build target 'stm32f103xb_if'
Before Build - User command #1: ..\..\..\tools\pre_build_script.bat
Python 3.11.5

f:\1\DAPLink-main\projectfiles\uvision\stm32f103xb_if>python ..\..\..\tools\pre_build_script.py 
fatal: Not a valid object name HEAD
#> Getting git description
#> ERROR: Failed to get git description, do you have git.exe in your PATH environment variable?
".\build\stm32f103xb_if.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:00

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & c:\Users\LONG3\.vscode\extensions\cl.keil-assistant-1.7.0\bin\Uv4Caller.exe -o f:\1\DAPLink-main\projectfiles\uvision\stm32f103xb_if\.vscode\uv4.log --uv4Path 'D:\keil5 mdk\core\UV4\UV4.exe' --prjPath f:\1\DAPLink-main\projectfiles\uvision\stm32f103xb_if\stm32f103xb_if.uvprojx --targetName stm32f103xb_if -c '${uv4Path} -b ${prjPath} -j0 -t ${targetName}'”已终止，退出代码: 1。 
 *  终端将被任务重用，按任意键关闭。 

I am the same, but I don’t know how to disable python. Please solve this problem. It has been bothering me for a long time.

0xc0170, 2023-10-09:
@LONG300002 I assume this is again about git not python, clone the repo instead of downloading zip (you can verify it by running git command like `git status` to check if the repo is in correct state).