ARMmbed/DAPLink #649: DAPLink can't be detected in MDK-ARM V5

ChinaXiaoMao, 2019-09-07:
I made a DAPLink using STM32F103. It can't be detected in MDK-ARM V5. When I open the setting of CMSIS-DAP in Keil V5, the LED turn on a short time but no debug unit device found. I get these message in Bus Hound
[image]
But it can be detected in EWARM V8. When I start a debug sesson in EWARM, I get these message in Bus Hound
[image]

0xc0170, 2019-09-09:
What version of MDK? What about pyOCD also having issues?

ChinaXiaoMao, 2019-09-10:
> What version of MDK? What about pyOCD also having issues?

5.28a

flit, 2019-09-10:
There is a known issue with MDK where it does not detect a CMSIS-DAP interface with interface number greater than 0.

0xc0170, 2019-09-10:
It has not yet been fixed in MDK, we will follow up on this issue with MDK team. Thanks @flit !

ChinaXiaoMao, 2019-09-19:
@0xc0170 
I tried stm32f103xb_stm32f103rb_if and it works. Maybe only stm32f103xb_if doesn't work.

0xc0170, 2019-09-19:
These two differ in the records (what files are included). There was this recent change https://github.com/ARMmbed/DAPLink/pull/629 that could make the difference for `stm32f103xb_if `

flit, 2019-09-25:
@ChinaXiaoMao The issue in MDK has been fixed. Please see this Keil knowledge base article with an attached updated CMSIS-DAP driver DLL: http://www.keil.com/support/docs/4170.htm. An

0xc0170, 2019-09-26:
I'll close this as resolved.

closed 2019-09-26 by 0xc0170