ARMmbed/DAPLink #650: Does the DAPLink support debugging risc-v ? 😅

balanceTWK, 2019-09-07:
Need to modify the source code to support risc-v jtag ?  囧rz

🤣

40Grit, 2019-09-07:
For a giga devices part?

Does risc-v even have standard debug silicon yet? Particularly for MCU's?

balanceTWK, 2019-09-07:
> For a giga devices part?
> 
> Does risc-v even have standard debug silicon yet? Particularly for MCU's?

yes ,giga devices.

I have a development board. ( GD32VF103 RISC-V 32-bit MCU evaluation Board ) there is a gd-link on the board and I found that gd-link was identified as cmsis-dap.

flit, 2019-09-07:
Sorry, DAPLink doesn't support RISC-V.

Very interesting that gd-link identifies as CMSIS-DAP. Do you mean the USB device product name includes "CMSIS-DAP"?

40Grit, 2019-09-07:
additionally what part number is the gd-link controller?

balanceTWK, 2019-09-08:
I'm just detecting it with MDK

[image]

closed 2019-09-08 by balanceTWK

0xc0170, 2019-09-09:
> Very interesting that gd-link identifies as CMSIS-DAP. Do you mean the USB device product name includes "CMSIS-DAP"?

From their documentation : "GD-Link adapter is a three-in-one multi-function development tool for GD32 series of MCUs. It provides CMSIS-DAP debugger port with JTAG/SWD interface".

I assume this is the debug target; https://github.com/ARMmbed/DAPLink/pull/538 ?

@balanceTWK Before you ask a question, look at documentation and source code, you can find your answer there 😉