ARMmbed/DAPLink #651: Add SWO support for the k20dx HIC

millerc-ep, 2019-09-13:
It appears that there is support for this in the k26 target via software, but not for the k20. We're looking to see if/how SWO support for the k20 is possible.

@maclobdell @brianesquilona @loverdeg-ep @trowbridgec

flit, 2019-09-13:
It's certainly possible from a technical standpoint. But unfortunately, none of the boards I know of  that use the K20 (i.e., FRDM boards) have the SWO signal from the target connected to the K20.

loverdeg-ep, 2019-09-13:
Good to hear.

@flit Our Flidor Development Adapter connects SWO.

Currently our Agora target is configured to use it.
https://os.mbed.com/platforms/agora-dev/

https://github.com/ARMmbed/DAPLink/blob/master/source/board/ep_agora.c

Is there anything we should do to indicate the actual debug adapter is Flidor versus just calling the combination of Flidor and Agora, "ep_agora".

loverdeg-ep, 2019-09-13:
[image]

[image]

flit, 2019-09-13:
Nice!

I would just keep the name ep_agora since Flidor is (at least currently?) only used with Agora. Or you could name it "ep_flidor_agora".

The main requirement for enabling SWO is a CMSIS-USART driver. You can probably just copy the one from the K26F, plus the required SDK UART driver.

Then configure the `SWO_*` macros in the K20 HIC's `DAP_config.h` file. `SWO_UART` should be set to 1, `SWO_BUFFER_SIZE` based on free RAM, might need to adjust `SWO_UART_MAX_BAUDRATE` since the K20 has a max clock speed of 50 MHz.

(Note: I had trouble using the DMA-based CMSIS-USART driver, just never got it to work and didn't have time to debug. But the `fsl_uart_edma.{c,h}` driver and EDMA driver are still in the K26F HIC folder. Would like to get that working some time. It might be more important for the K20 since it doesn't have as many cycles to spare as the K26F running at 120 MHz.)

mbrossard, 2021-06-11:
`lpc55xx` was added with SWO support and it is being added to `lpc4322` in #813 (which supersedes #712). In the pull request you can see that `SWO_UART` is defined to 0 by default but can set in the board record file.

@millerc-ep, @loverdeg: Are you still interested in doing this?