ARMmbed/DAPLink #652: Disallow both CMSIS-DAP v1 and v2 to be opened simultaneously

flit, 2019-09-13:
In interface firmware configurations where both CMSIS-DAP v1 and v2 USB interfaces are available, there is a possibility that the user could open both CMSIS-DAP interfaces concurrently. This will, of course, cause unpredictable behaviour.

A real world example is with multicore debug using Keil µVision. If starting a debug session in one µVision instance with the CMSIS-DAP v1 interface, and the CMSIS-DAP v2 interface is selected in the setup dialog of another µVision instance, then the debug connection in the first is lost. The same happens the other way around (debug session with CMSIS-DAPv2 and setup dialog with CMSIS-DAPv1).

The solution is to disallow simultaneous opening of both CMSIS-DAP interfaces. The actual mechanism used to disable a CMSIS-DAP interface needs to be investigated.

loverdeg-ep, 2019-09-13:
I have always sort of wished that DAPLink could expose another usb->com port specifically for exposing a CLI to configure the probe. Yet another usb endpoint though.

Not a fan of serial escape sequences or I'd say use that as an alternative.

flit, 2019-09-13:
A CLI is an interesting idea. But I think the drag-and-drop configuration files are probably easier for most people. And they become a sort of command line interface when activated with commands like `touch /Volumes/FRDM-K64FD/auto_rst.cfg`.

Do multiple CDC virtual com ports require additional USB interfaces? I don't know CDC well enough to know the answer to that.

loverdeg-ep, 2019-09-13:
@aglass0fmilk ^ 

> Do multiple CDC virtual com ports require additional USB interfaces? I don't know CDC well enough to know the answer to that.

RadioOperator, 2019-09-23:
I think, multi DAP in system, what ever two DAP or 1-DAP + 1-JLINK , in different USB ports. This case needs the IDE supported.  Suggest you try 1-DAP + 1-JLINK, select by different IDE setup to see what happened. In com port case, we have com port number to define the Com channel, DAP device does not have this.