ARMmbed/DAPLink #653: CRC test failed

flymdkone, 2019-09-16:
CRC test failed, please help me

`
After Build - User command #1: ..\..\..\tools\post_build_script.bat C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if C:\Keil_v5\ARM
Python 3.7.4
C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if>fromelf --bin C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if.axf -o C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if.bin 
C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if>fromelf --i32 C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if.axf -o C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if.hex 
C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if>python ..\..\..\tools\post_build_script.py C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if.hex C:\Users\ASUS\Desktop\DAPLink\projectfiles\uvision\stm32f103xb_stm32f072rb_if\build\stm32f103xb_stm32f072rb_if_crc 
  File "..\..\..\tools\post_build_script.py", line 144
    pack_flash_algo.symbols['EraseChip'] + blob_header_size + flash_blob_entry if pack_flash_algo.symbols['EraseChip'] != 0xffffffffL else 0,
                                                                                                                                    ^
SyntaxError: invalid syntax
".\build\stm32f103xb_stm32f072rb_if.axf" - 1 Error(s), 0 Warning(s).
`

0xc0170, 2019-09-16:
Please use python 2.x at the moment, we need to fix this to get python 3 supported

0xc0170, 2019-09-18:
Does that fix your issue?

flymdkone, 2019-09-19:
> 这会解决您的问题吗？

no，More problems with python2.x, pip installation error！
I don't need a CRC test line? Then I can use python3.7.4！！！

0xc0170, 2019-09-19:
I don't understand what python 2 problems you are having?

crc is required.

0xc0170, 2019-10-24:
Please reopen with more details

closed 2019-10-24 by 0xc0170

0xc0170, 2019-11-07:
I was able to reproduce this error with python 3 though. This repository is not yet compatible with python 3, I'll create an issue and would like to migrate.