ARMmbed/DAPLink #654: #include does not have the correct case

lwalkera, 2019-09-16:
https://github.com/ARMmbed/DAPLink/blob/master/source/rtos_none/SysTick_Handler.c#L22 Fails to compile on case-sensitive systems because Systick_Handler.h != SysTick_handler.h.

0xc0170, 2019-09-18:
Looking at the files: https://github.com/ARMmbed/DAPLink/tree/master/source/rtos_none , I don't see any mistake there.

`SysTick_handler.h` is not included but proper `SysTick_Handler.h` is also in the files above.

We test mac OS/win/linux so should have seen these previously?

lwalkera, 2019-09-19:
The #define is "Systick_Handler.h" it should be "Sys**T**ick_Handler.h"

0xc0170, 2019-09-19:
Now spotted ! Please send PR fixing the issue

0xc0170, 2019-10-21:
I'll fix it later today

flit, 2019-10-21:
@0xc0170 I think the reason this wasn't seen before is that CI only builds on Windows, then tests on the 3 OSes. Before mbed-cli based building was supported, building on Windows was the only option because it required Keil MDK. Now we do have the option to build on all 3 OSes, assuming we had an AC5 compiler.

commit 4cfd564c (2019-10-21): rtos: fix systick inclusion

Fixes #654
PR #661 (2019-10-21): rtos: fix systick inclusion
Fixes #654
closed 2019-10-21 by 0xc0170