ARMmbed/DAPLink #655: Red Bear Lab BLE Nano version 1.0 binaries available, but not v1.5

linuxtim, 2019-09-17:
Thanks for making available binaries.  You list the Red Bear Lab BLE Nano at https://armmbed.github.io/DAPLink/ as "RedBearLab-BLE-Nano" , but this support is (I think) only for the (NXP based) programmer included in the version 1 kit.  The version 2 is based on an STM32 instead.

Could you make it clear that the supplied binary is for v1 only?

It would be great if you could also make available precompiled binaries the v2 kit ("RedBearLab-Daplink-v1.5" and "ReadBearLab-BLE-Nano-v2" maybe?).

https://redbear.cc/product/retired/ble-nano-kit.html
https://redbear.cc/product/retired/ble-nano-kit-2.html

0xc0170, 2019-09-18:
Don't they provide for bot of these kits the same daplink:: " This Kit include a companion board, DAPLink v1.5, which is used to load firmware into BLE Nano from a PC."? This is the readme for their daplink: https://github.com/redbear/nRF5x/tree/master/USB-IF

0xc0170, 2019-10-21:
Any update to the question above?

linuxtim, 2019-10-21:
Hi, sorry I'm not sure I understand your question.  Also I think my original bug report wasn't clear...  The new owners of Red Bear don't seem to want to support these, and most of the Red Bear infrastructure seems to be disappearing...  Red Bear made an external programmer "MK20 USB v1.0" - based on an stm32f103 with 128K flash, which is capable of running Daplink (unfortunately also there is a Daplink "MK20 USB v1.5" as well, but this doesn't have sufficient flash for DAPLink).

For the "MK20 USB v1.0", the version of the DAPLink binary offered is now old, and probably won't be updated, so it would be great if it could be built and made available at https://armmbed.github.io/DAPLink/ - but perhaps this is too rare/niche hardware to worry about supporting at this stage.  I think it would perhaps be more useful if binaries were made available for common STM32F103CBT6 based programmers e.g. ST Nucleos etc.  Many of the third party programmers use the same pinout (including the "MK20 USB v1.0" I think - I don't have one to hand at the moment).

So perhaps it's better to close this bug and open a more general one about making binaries available for stm32f103CBT6 boards...

closed 2019-10-21 by linuxtim