ARMmbed/DAPLink #656: Add new HIC stm32f723 request

RadioOperator, 2019-09-23:
Dear Sir,
The famous stm32f103xb in DAPLink very long time, but it's only USB full-speed capability and limited Flash. Recently, I have finished 1st test version CMSIS-DAP v1 & v2 on STLINK-V3MINI device. Refer to 
https://github.com/RadioOperator/CMSIS-DAP_for_STLINK-V3MINI
It's a new lower cost USB 2.0 high speed hardware board, everything ready to go DAPLink.

Would you help to port the STM32F723 into DAPLink HIC list?  or any tips.

0xc0170, 2019-09-23:
> Would you help to port the STM32F723 into DAPLink HIC list? or any tips.

If you can be more specific, we might help you.

RadioOperator, 2019-09-23:
0xc0170, I'm trying to port the IC interface drivers based on stm32f103xb. With your supporting, would be successful definitely. The STM32F723 has 216MHz core Freq. and USB high-speed, would be the best DAPLink device. I'm not full-time to do this, please wait. If I got something cannot solve by myself, I will let you know. Thanks.

40Grit, 2019-09-23:
@aglass0fmilk

RadioOperator, 2019-10-19:
Hi, 0x0170, because the latest STM32F7 series USB driver worked for RTX5/CMSIS and ARM RL-USB (2019), according to the info in https://github.com/ARMmbed/DAPLink/issues/197, I think it's better to wait the official RTX5 DAPLink release, then add STM32F723 as new HIC should be very easy, I hope the new release shares the same API with the latest version RL-USB and Keil/MDK-Middleware.

closed 2019-10-22 by RadioOperator

vincent290587, 2024-02-09:
Hi ! Seems like this issue could be re-opened ?

mathias-arm, 2024-02-10:
Hi @vincent290587 are you (or @RadioOperator?) interested in working on this project? Since the issue was closed a lot of things have progressed: DAPLink was ported to GCC and RTX5. The main hurdle remaining is the USB driver but there are resources available:
 - Keil opened the drivers for [STM32F4xx](https://github.com/ARMmbed/DAPLink/tree/develop/source/hic_hal/stm32/drivers) and that was used as a basis for the [`stm32h743xx` HIC](https://github.com/ARMmbed/DAPLink/blob/develop/source/hic_hal/stm32/stm32h743xx).
- The CMSIS USB drivers (`CMSIS/Driver/USBD_HS_STM32F7xx.c`) in the Keil `STM32F7xx_DFP` CMSIS pack, but it is designed for a newer driver API, so it can't work as-is.

For anyone interested in working on this, I can point to a branch that has a lot of the foundational work, including some partially working USB driver (at Full-Speed, no High-Speed).

mathias-arm, 2024-02-27:
I have a branch with multiple STM32 targets (they all share the same USB driver for USB OTG): https://github.com/mbrossard/DAPLink/tree/feature/stm32_otg_usb

vincent290587, 2024-03-21:
I have started some work to port DAPLink to STM32F723, but could you answer some questions ?

- how did you generate the flash_blob.c ?
- Which USB drivers are you using ? (I am using those in the hic_hal/stm32/shared folder)
- I managed to get the bootloader to a point where there is "some" USB activity (until I tget invalid USB BOS descriptor), could you advise on USB debugging tools ?

In any case many thanks for the code