ARMmbed/DAPLink #657: CDC serial transmission corrupted after resuming debugging session

mhanak, 2019-09-24:
The DAPLink virtual serial port operation gets corrupted when more than 8 characters are sent from host PC to target MCU while it is stopped during a debugger session. After resuming the execution, the virtual serial port does not properly transmit the characters being generated by the MCU.

The issue can be reproduced with a basic terminal echo application:

Steps to reproduce:
1. Have a UART echo application running on a target MCU under a debugger session.
2. Connect with TeraTerm or other terminal application on host PC to target DAPLink CDC port.
3. Type the characters to test proper operation. Characters are echoed back and displayed in the terminal.
4. Break/suspend the MCU application execution in in the debugger.
5. Type more than 8 characters in the terminal application, characters are not echoed obviously.
6. Resume the MCU application execution in the debugger. The first 8 characters previously typed will be echoed - sometimes in a wrong order.
7. Continue to type characters in the terminal application, the characters are NOT echoed properly anymore. The character echo is delayed by one or more typed characters.

Observations:
- Closing/opening the virtual COM port on host PC does not help
- Disabling/enabling the virtual COM port device on host PC does not help
- Re-starting the debugger session does not help
- The only way how to bring the port back to normal is to unplug/plug the USB cable.
- Similar behavior observed also with JLink firmware running on K20 OpenSDA interface.

Tested using NXP MCUXpresso and FRDM-K64F board and basic terminal echo application (hello world demo from MCUXpresso SDK suite). 

Tried with interface version 0246 and the latest 0253_K20DX_FRDMK64F_0X5000.BIN

0xc0170, 2019-10-21:
@mhanak Thanks for reporting. 

I've checked some source files. I found one line that might be causing this behavior, see https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/usb2uart/usbd_user_cdc_acm.c#L178 .  The check does not look correct to me, I'll send a fix. This would explain why it only transfers 8 characters (it's always caped to 8 chars). The buffer should be 64 characters long.

Can you try to rebuild daplink with fixing this check if it helps?

0xc0170, 2019-10-21:
@mhanak Thanks for reporting.

I'll investigate in the following days hopefully.

mhanak, 2019-10-22:
Hi Martin, I'm sorry I cannot rebuild now, but I can test a binary if you
send me one.

0xc0170, 2019-10-22:
No problem, once I have something will share.

My initial findings were incorrect, I'll discuss this bug with @c1728p9

mhanak, 2019-10-23:
Hello, you were right, resetting the K64F helps - which means this is the issue of internal UART module operation after resuming the debugger session. I will investigate more and will close this issue as soon as I confirm it. Thanks!

mhanak, 2019-10-24:
Yes, I have confirmed directly on the UART TX pin that the DAPlink sends correct data bytes. The corrupted character echo must be caused by K64F UART module being in wrong state after resuming the execution. 

No issue in DAPlink. 
I'm closing this ticket.

Thank you for your advice.

closed 2019-10-24 by mhanak