ARMmbed/DAPLink #658: NRF52840_DK intermittently does not mount after power cycle

OPpuolitaival, 2019-09-25:
Occurs with daplink v0254. Not able to reproduce with v0253.

The issue frequently reproduces with the help of ykush power switch.
```
$ ykushcmd -d a && sleep 0.2 && ykushcmd -u a && sleep 5 && mbedls -u

| platform_name | platform_name_unique | mount_point | serial_port | target_id | daplink_version |
|---------------|----------------------|-------------|--------------|--------------------------------------------------|-----------------|
| NRF52840_DK | NRF52840_DK[0] | unknown | /dev/ttyACM0 | 1102360250203120354e4351313034203031303597969903 | unknown |
```

Failure case dmesg log:
```
[  +2.239612] usb 1-1.7.1: new high-speed USB device number 67 using xhci_hcd
[  +0.101551] usb 1-1.7.1: New USB device found, idVendor=0d28, idProduct=0204
[  +0.002676] usb 1-1.7.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.002632] usb 1-1.7.1: Product: DAPLink CMSIS-DAP
[  +0.001009] usb 1-1.7.1: Manufacturer: ARM
[  +0.000773] usb 1-1.7.1: SerialNumber: 1102360250203120354e4351313034203031303597969903
[  +0.001381] usb-storage 1-1.7.1:1.0: USB Mass Storage device detected
[  +0.001301] scsi host3: usb-storage 1-1.7.1:1.0
[  +0.001186] cdc_acm 1-1.7.1:1.1: ttyACM0: USB ACM device
[  +0.002303] hid-generic 0003:0D28:0204.00CC: hiddev0,hidraw0: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-0000:00:14.0-1.7.1/input3
[  +1.018393] scsi 3:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
[  +0.004578] sd 3:0:0:0: Attached scsi generic sg1 type 0
[  +0.001048] sd 3:0:0:0: [sdd] Attached SCSI removable disk
```

Internal id for this issue: MBEDOSTEST-723

0xc0170, 2019-09-26:
I'll review these , what has changed between the firmware to cause this error .

0xc0170, 2019-11-28:
@flit @brianesquilona Any guess what could break between these 2 releases?

@OPpuolitaival We need better way to reproduce this. I don't have a switch like that, to break the device locally.

0xc0170, 2019-11-28:
Lets use 0253 in the meantime and find out what has changed that could cause this