ARMmbed/DAPLink #659: EFM32 Support?

pfried, 2019-10-14:
@brianesquilona 

I saw your code commits concerning EFM32GG support fort he DAPLink:

https://github.com/ARMmbed/DAPLink/blob/25cbc71260c318e15f9c459f39a8fc27641de92d/source/family/siliconlabs/efm32gg/flash_blob.c#L24-L47

We are currently using Silicon Labs EFM32 chips in our products. Has Silicon Labs shown no effort to provide DAPLink support for their cores or why did you do the implementation? We are currently forced to use a matching STK board for programming those chips. (Segger JLink OB)

@stevew817 Would you be able to provide DAPLink support for the EFM32 targets? Or could you help with an implmentation?

Best Friedrich

stevew817, 2019-10-14:
@pfried If you look at the git blame, it was me who pushed the EFM32GG flashloader to DAPLINK. I don't think @brianesquilona has something to do with it?

EFM32GG support was added into daplink for a special project at the time, which regrettably got canceled later. Because of that cancellation, we had no-one asking for daplink integration anymore, and decided to not pursue it because of bandwidth constraints on the engineering team.

What's the specific use-case you want to enable here? How would having daplink improve your setup over using JLink? If this information is sensitive, feel free to contact me directly.

pfried, 2019-10-14:
We are currently building up a development chain for mbedOS, currently using EFM32 targets. We want to use the following setup:

* mbedOS Studio (We have a working Eclipse + JLink GDB + mbedCLI setup, but mbedOS Studio is far more integrated and we dont need to maintain a "custom" development environment.
* DAPLink for programming, debugging and serial connection
* EFM32 targets (various GGS1, PGS1, PG, ...)

We have that solution for our Nordic Semi based solutions and it is really great to work with.

flit, 2019-10-14:
Is it DAPLink's mass storage flash programming that is not working with the EFM32 devices?

Usually if mass storage flash programming doesn't work with a particular target, it's because that target requires special handling of reset or reset and halt. Often a ROM bootloader gets in the way.

cederom, 2019-10-14:
If the problem is only with Drag-n-Drop flashing while CMSIS-DAP endpoint works fine you can always work and flash with https://github.com/mbedmicro/pyOCD :-) `pyOCD` has this nice utility called `flashtool` (as standalone or commandline parameter) that is even faster than Drag-n-Drop and does not overwrite memory regions with the contents that is already there so it better against flash wear-out :-)

pfried, 2019-10-14:
The EFM32 use a [JLink OB](https://www.segger.com/products/debug-probes/j-link/models/j-link-ob/) which provides a drag&drop interface. This does not work with the mbedStudio, but if you drop a `.hex` file onto it it works for the devboard itself, but if you want to use the devboard as a programmer for an external target problems start (We found out you can use the devboard with the same chip as the target).

The target detection within mbedStudio is somehow unclear to me, does it work with `mbedls` or is it using PyOCDs capabilities?

We do have  a somehow working solution in place, but ideally we would have something working with mbedStudio even if that means selecting a different controller for our future boards

cederom, 2019-10-14:
Drag-n-Drop interface is provided by DAPLink as whole thing. It shows up as mass storage device and is handled fully by the DAPLink itself when you write a hex file onto.

External chip problems, although Target is the same chip, seems to be caused by Reset signalling as mentioned by @flit. Some chips use Hardware some use Software Reset + Chip Init.

DAPLink has hardcoded Target FlashAlgo for only one given specific target so you cannod use that device with a different Targets for UMS flashing. That particular DAPLink firmware will only provide Drag-n-Drop for the Target family that is mounted on the board. Still you can use CMSIS-DAP endpoint to use given HIC/DAPLink to work with other Targets.

mbedStudio (probably?) detects target based on TXT file that is stored on a mass storage device because that works even when you do not have mbed-cli installed. Please correct me if I am wrong?

Please check the RESET signalling and play with pyOCD with that external target - that could reveal some important hints for you :-) You may also want to use CDC / Virtual-Serial-Port endpoint to trigger Reset on that external Target and see if that works.

pfried, 2019-10-14:
Yes DAPLink is one thing, EFM32s do not have a DAPLink support since the DAPLink firmware needs the flash blob (FlashAlgo) and other hardware description (addresses etc.). We are aware of the reset issue.

The EFM32s do have a JLink which **is not** a DAPLink derived from this repository at least what I can tell. They provide a similar drag&drop feature and interface so it somehow should be compatible but mbedStudio does not recognize it.

Since PyOCD has CMSIS Pack support for EFM32s I will try to get that running, but how targets are detected by mbedStudio I dont kow. Maybe its the bundled [mbedls](https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-ls) or some other mechanism. As the development does not happen on Github I cannot take a look at the source. I have another open [Issue at the mbed-os-tools repository](https://github.com/ARMmbed/mbed-os-tools/issues/201)

The best thing that could happen is a open source programmer / debugger based on this firmware which is able to work with all the mbedos targets and works with mbedStudio which would free us from the non-working programming problems.

cederom, 2019-10-14:
> The best thing that could happen is a open source programmer / debugger based on this firmware which is able to work with all the mbedos targets and works with mbedStudio which would free us from the non-working programming problems.

You are looking for pyOCD :-)

flit, 2019-10-14:
Mbed Studio uses mbedls to detect targets, I believe. cc @thegecko 

I believe a user has successfully used the EFM32 CMSIS-Packs with pyOCD.

thegecko, 2019-10-14:
> The target detection within mbedStudio is somehow unclear to me

Mbed Studio detects targets using known vendorIDs. Depending on the manufacturer, different tools are interrogated to determine the deploy and debug support available. These include:

- PyOCD's database
- The CMSIS packs we distribute with Mbed Studio (stripped to be only debug info)
- mbed-ls (only if PyOCD doesn't support the target and in some other special cases)

If a target is supported by PyOCD, that used used to run the application. Otherwise MSD copy is instigated.
If a target is supported by PyOCD or has a CMSIS pack, debug is enabled.

>  ...open Issue at the mbed-os-tools repository

This repo is for the OSS build system and mbed-cli. As Mbed Studio isn't open source, you will likely get a better response through the feedback tool in the product or the Mbed forums.

> take a look at https://github.com/ARMmbed/dapjs

This was built for accessing targets from the browser.

Regarding this situation, we are currently supporting Mbed enabled targets in Mbed Studio initially. We are considering how custom targets can be supported in future (which I believe is the setup you have here, @pfried ), so would be keen to understand your use case a little more.

pfried, 2019-10-15:
@thegecko Thank you for the explanation of the target detection.

Our use case is actually simple: We do electronics development for our customers and we want to base an inhouse development on mbed and are currently figuring out how to do this in a proper way. Historically coming from Silicon Labs we are currently using one of their chips (EFM32PG Series 1).

Creating a target within mbedOS is quite easy as we can branch from existing EFM32 targets. So compiling with mbed-cli or even mbedStudio works (in the lastest version you can select custom targets within the select box).

For programming and debugging we want to use a DAPLink (ideally one programmer for all developments). We even have a custom board with one onboard (no EFM32 target, but Nordic).

     PC: mbedStudio  -->  DAPLink  -->  Custom EFM32 Target 

So the task would be: Tell mbedStudio that target XY is available for programming / debugging. My idea was to trick mbedls to mock a custom target, but no success so far.

From a workflow perspective it would be great to have a file within the project (like the mbedls.json) to do the mocking, for the development you could then just use a per develope DAPLink which shows up as the correct target for each project as long as the DAPLink does not change.

@flit Yes, I opened the issue by the time: https://github.com/mbedmicro/pyOCD/issues/280 and someone reported success

I will try to get that working as well, so the setup would be:

     PC: pyOCD  -->  DAPLink  -->  Custom EFM32 Target

Is that correct, and the command would be something like: `pyocd gdbserver --pack ./Keil.EFM32Gxxx_DFP.1.4.0.pack -t efm32gg990f1024` Are my assumptions correct?

thegecko, 2019-10-15:
We are looking into a feature in Mbed Studio where an unknown target which has been connected can then be assigned a target from the drop-down. That should partially meet your needs I think, but debugging is unlikely to work until we ship the CMSIS pack you need.

pfried, 2019-10-15:
Any chance to add a pack manually?

@flit The pyocd debugging server seems to work with: `pyocd gdbserver --pack ./SiliconLabs.EFM32PG12B_DFP.5.8.2.pack --target=EFM32PG12B500F1024GM48`

thegecko, 2019-10-15:
> Any chance to add a pack manually?

Not at this time. We are able to push out further packs as we add them to the system, could you create a request to add the pack you need via the Mbed Studio feedback link?

flit, 2019-10-15:
@pfried Glad to hear pyOCD debugging works! You can also install EFM32 support via managed packs: `pyocd pack --install efm32pg12`. Then you don't need to pass the `--pack` argument on the command line.

0xc0170, 2019-10-21:
@pfried Was all your questions answered, or is there anything left for this issue (besides the new feature for Mbed Studio).

pfried, 2019-10-23:
@0xc0170 Hoping to see some mbedStudio support soon, but for now we can close it. Thanks

closed 2019-10-23 by pfried