ARMmbed/DAPLink #663: NXP i.MX RT Hic?

raveslave, 2019-10-25:
The new iMXRT1062 would be a great addition as a debug-adapter.
- Fairly low cost
- High speed USB
- on-board flash that could be used to store a rom "locally" to achieve high speed flashing in factory setups

The Teensy4 board could be used as a nice & compact debug adapter

Anyone willing to add support for this CPU?

loverdeg-ep, 2019-10-25:
I was thinking the same thing at bit back.

The peripherals for i.MX RT series are particularly complex and might be a little overkill but pricing and clock speed is great for these parts.

Have you considered the LPC55 series? Dual core M33 (Potential secure debug applications), I believe it has some logic blocks too.

raveslave, 2019-10-25:
LPC55S6x are great. A cool feature is the built-in "programmable logic unit", this could be used to speed up SWD a lot! Also great with it's On-chip ROM bootloader, so no need to use a swd-adapter to flash your daplink board once built.

Note that the power-supply design using the RT will be more simple thanks to it's integrated DC-DC converter.

flit, 2019-10-28:
We are planning to add LPC55xx support at some point (soon, hopefully).

raveslave, 2021-02-28:
checking in, any "plans" :)

flit, 2021-02-28:
Not for an RT series HIC (it's overkill imo), but the LPC55S69 port is basically done: [feature/lpc55xx_hic](/flit/pyOCD/tree/feature/lpc55xx_hic) on my fork.

This is based on the `experimental_compilers` branch, so it supports gcc (I only tested the gcc-built version). It will be merged upstream once the `experimental_compilers` branch is merged in, after testing and debugging is complete. Which will take a while due to the number of supported HICs. Though I guess we could merge the LPC55xx port into `experimental_compilers`.

I also implemented DMA-based SWD on the [feature/lpc55xx_swd_dma](/flit/pyOCD/tree/feature/lpc55xx_swd_dma) branch. Unfortunately I ran into an issue where DMA triggers from the timer seem to get lost for no apparent reason, feels like a hardware issue but not sure. I've reached out to NXP but haven't heard back yet.

raveslave, 2021-02-28:
Just about to design a board and looking at 
- LPC4322 (used as debugger on most nxp ev-kits)
- LPC4370 (used on the LPC-Link2 board) 
  * _didn't see the hic for that mcu in here though, any clues where nxp keeps this!?_ 
- LPC55S69
  * what would you say are the 'pros' over the LPC43xx from a functional perspective?
  * any ideas on when this might be merged?

---   
ps. nice with SWD+DMA for speed, NXP should be able to comment on that.

40Grit, 2021-02-28:
I can confirm @flit 's branch for lpc55 works as tested with a custom lpc55 hic.

Only issue now is lpc55 availability.

raveslave, 2021-03-02:
right, thanks for info. I got one of NXP's new McuLink boards, it uses the LPC55S69     
afaik, they just provide a binary at this stage.. does anyone know if this is based on a certain branch here?

flit, 2021-03-02:
The NXP MCU-Link firmware is based on their LPC-Link2 firmware, which is closed source. (So the LPC4370 firmware is also closed.) It's not related to DAPLink, other than also making use of the CMSIS-DAP reference code.

raveslave, 2021-03-05:
less good that they keep it closed source, especially since they use the pyocd interface to run it

flit, 2021-03-08:
Well, DAPLink works on the MCU-Link circuit. So the fact that NXP's firmware is closed source is mostly an annoyance rather than a serious problem. The most important thing is that the circuit is open.

For a new board, I'd generally recommend the LPC55S69 since it's newer. But there are problems with availability and long lead times. So the LPC4322 may be a better option depending on your schedule.

Differences:

**Feature** | **LPC55S69** | **LPC4322**
----|---|---
smallest package | VFBGA98 7x7 | TFBGA100 9x9
smallest QFP package | HTQFP64 10x10 | LQFP144 20x20
internal flash | 640 kB | 512 kB
internal SRAM | 104 kB | 320 kB
HS USB | yes | yes
max CPU clock | 150 MHz | 204 MHz
main core | M33 | M4
2nd core | M33 | M0
1k Mouser price |  <$3.82 (only showed price for 520) | $7.99

There have been issues on some LPC4322 devices with running at the full 204 MHz clock; we were told internally to not run at that speed in the DAPLink port. Not sure what the full details are.

The LPC4322 does have the Serial GPIO (SGPIO) peripheral which looks really interesting for performing high speed SWD/JTAG.

There are other devices in the LPC55xx family such as the LPC5528 that look nice and are less expensive. Mouser shows $3.10/2000pcs for LPC5528JEV98Y.

raveslave, 2021-03-11:
thx for the info @flit 

more SRAM in 4322, I guess not neccessary since we have high-speed usb on both, so no need for large fifo's 
SGPIO is interesting, what was your impl./idea here?  

I remember seeing some older pyocd/daplink project where the developer made use of SPI "SISO" to achieve 'segger speeds' 
by speeding up the swdio pipe.

feature-wise, does both LPC55S69 & LPC4322 provide the same feature-set and stability!?
I noted that NXP actually uses LPC43**70** on mcu/lpclink2

raveslave, 2021-03-18:
@40Grit / all, what board did you use to run the LPC55S69 version on?
looking at making my own formfactor based on MCU-Link-revPA6

raveslave, 2021-03-18:
re availability, LPC55S66 seems easier to find than LPC55S69, but has less ram and flash...
144 KB SRAM vs 320 KB / 256 KB Flash vs 640 KB
any ideas if that would be sufficient?

40Grit, 2021-03-18:
Ran on both the nxp reference board as well as a custom design.

40Grit, 2021-03-18:
@flit can comment on whether a lower spec lpc55 would suffice

flit, 2021-03-18:
@raveslave

There's a reason I called the HIC "LPC55xx" in the DAPLink code. 😉 

The RAM and flash should be sufficient. In the current `lpc55xx_if` build, flash is 76028 bytes and RAM is 24664 bytes. Plenty of room.

You might need a different [`daplink_addr.h`](https://github.com/flit/DAPLink/blob/feature/lpc55xx_hic/source/hic_hal/nxp/lpc55xx/daplink_addr.h) file. Although I currently have the memory region sizes set to less than maximum; we can change it to fit the smaller LPC55xx devices by default. (DAPLink firmware images currently have to be fixed size, with padding out to the end of the allocated mem region; related to CRC, I have a todo to fix some time.)

Sorry I mixed up the LPC55S69 RAM size in the table above. It's fixed now.
SGPIO just looks useful for a high speed SWD without some of the complication of timer-triggered DMA. I don't have time to implement it, though.

Both LPC43xx and LPC55xx are equally stable afaik. Arm uses the LPC43xx family for probes like ULINKplus without any issues that I'm aware of.

(I'm going to transfer this issue to a discussion.)

closed 2021-03-18 by flit