ARMmbed/DAPLink #667: Invalid binary -> FAIL.txt "The transfer timed out."

psandelin, 2019-11-04:
When drag and dropping a nRF52 binary the first bytes are validated in to be valid.
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/validation.c#L55

The problem is if they are not valid, the FAIL.txt will contain 
"The transfer timed out."
There seems to be an 
else if( NNN){} 
missing here
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/vfs_manager.c#L798

0xc0170, 2019-11-04:
Can you please send a fix via a pull request?

psandelin, 2019-11-04:
If I could I would. 
I tried a fix, but broke lot's of other stuff.

All I see is that when a binary that was invalid addresses on the first bytes was dragged and dropped.
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/validation.c#L55
fails.

After a while the USB state machine times out and the FAIL.txt is filled with the 
"The transfer timed out"

But the validation.c is called several times, (even without drag and drop ), something with USB drive handling in Windows.
So it is not as simple as to just check that there.

This problem needs someone who understands the USB code connection to the Daplink loop better.