ARMmbed/DAPLink #669: ERROR: pyocd 0.16.0 has requirement pyyaml<5.0,>=4.2b1, but you'll have pyyaml 5.1.2 which is incompatible.

andrewc-arm, 2019-11-06:
During the installation, I came across following ERROR. Is this OK?
```
ERROR: pyocd 0.16.0 has requirement pyyaml<5.0,>=4.2b1, but you'll have pyyaml 5.1.2 which is incompatible.
```

The full log.
```
c:\Andrew\code\DAPLink>venv\Scripts\activate

(venv) c:\Andrew\code\DAPLink>pip install -r requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\af\62\c4\bc2136982e9d0b7c653a0ca03a3b514f272fa84ddc80951a80\project_generator-0.9.13-cp27-none-any.whl
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\d5\50\f2\352d103b2ea211726b88264507eac08bd010ffc3a61248ae02\mbed_ls-1.6.2-cp27-none-any.whl
Collecting pyserial
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting pyOCD==0.16.0
  Downloading https://files.pythonhosted.org/packages/e4/ef/8f73f2ea8c810bda55b5dbc1c188edb7b843d4b47cafa80bf6f4b80ed7e4/pyocd-0.16.0.tar.gz (1.1MB)
     |################################| 1.1MB 781kB/s
Collecting requests
  Downloading https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl (57kB)
     |################################| 61kB 1.4MB/s
Collecting intelhex
  Using cached https://files.pythonhosted.org/packages/bf/77/bf670318b3db325c71e2ac6a90b7bcfdf9fc739b7cf6aebb31715721623e/intelhex-2.2.1-py2.py3-none-any.whl
Collecting six
  Downloading https://files.pythonhosted.org/packages/65/26/32b8464df2a97e6dd1b656ed26b2c194606c16fe163c695a992b36c11cdf/six-1.13.0-py2.py3-none-any.whl
Collecting enum34
  Using cached https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\1f\a9\7e\d189b5030ee3a56f9b72c28281bb11d661b8ea312e28de08a5\pyusb-1.0.2-cp27-none-any.whl
Collecting mbed-cli==1.8.2
  Using cached https://files.pythonhosted.org/packages/95/b9/4634ae13e421873e852ad8dcb4445fab2375376eaf6cb73369dbdf4aeadb/mbed_cli-1.8.2-py2-none-any.whl
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\a0\b0\b2\4f80b9456b87abedbc0bf2d52235414c3467d8889be38dd472\bs4-0.0.1-cp27-none-any.whl
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/4f/47/3ce61b9a00d1cce9500cca7a88e9b7105a1f6434be9ceaa748e09835b367/numpy-1.16.5-cp27-cp27m-win32.whl (10.0MB)
     |################################| 10.0MB 4.3MB/s
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\55\2f\15\4ce6885a52f475de68c16f3116a94d2156d588390cdb6c507c\pyelftools-0.25-py2.py3-none-any.whl
Collecting fuzzywuzzy
  Using cached https://files.pythonhosted.org/packages/d8/f1/5a267addb30ab7eaa1beab2b9323073815da4551076554ecc890a3595ec9/fuzzywuzzy-0.17.0-py2.py3-none-any.whl
Collecting jsonschema
  Downloading https://files.pythonhosted.org/packages/ce/6c/888d7c3c1fce3974c88a01a6bc553528c99d3586e098eee23e8383dd11c3/jsonschema-3.1.1-py2.py3-none-any.whl (56kB)
     |################################| 61kB 1.3MB/s
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\84\91\d8\5ed32405147161af83ef353b8b4d5dc42f0b1d2d83f8368d9e\project_generator_definitions-0.2.38-cp27-none-any.whl
Collecting pyYAML
  Downloading https://files.pythonhosted.org/packages/59/8a/489e2bd945c44b16f098c3746685609e5635478b11fb505da2b9b6ae5641/PyYAML-5.1.2-cp27-cp27m-win32.whl (193kB)
     |################################| 194kB 2.4MB/s
Collecting argparse
  Using cached https://files.pythonhosted.org/packages/f2/94/3af39d34be01a24a6e65433d19e107099374224905f1e0cc6bbe1fd22a2f/argparse-1.4.0-py2.py3-none-any.whl
Collecting Jinja2
  Downloading https://files.pythonhosted.org/packages/65/e0/eb35e762802015cab1ccee04e8a277b03f1d8e53da3ec3106882ec42558b/Jinja2-2.10.3-py2.py3-none-any.whl (125kB)
     |################################| 133kB 2.9MB/s
Collecting xmltodict
  Downloading https://files.pythonhosted.org/packages/28/fd/30d5c1d3ac29ce229f6bdc40bbc20b28f716e8b363140c26eff19122d8a5/xmltodict-0.12.0-py2.py3-none-any.whl
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\80\34\1c\3967380d9676d162cb59513bd9dc862d0584e045a162095606\prettytable-0.7.2-cp27-none-any.whl
Collecting appdirs>=1.4
  Using cached https://files.pythonhosted.org/packages/56/eb/810e700ed1349edde4cbdc1b2a21e28cdf115f9faf263f6bbf8447c1abf3/appdirs-1.4.3-py2.py3-none-any.whl
Collecting fasteners
  Downloading https://files.pythonhosted.org/packages/18/bd/55eb2d6397b9c0e263af9d091ebdb756b15756029b3cededf6461481bc63/fasteners-0.15-py2.py3-none-any.whl
Collecting future
  Downloading https://files.pythonhosted.org/packages/45/0b/38b06fd9b92dc2b68d58b75f900e97884c45bedd2ff83203d933cf5851c9/future-0.18.2.tar.gz (829kB)
     |################################| 829kB 2.3MB/s
Collecting websocket-client
  Downloading https://files.pythonhosted.org/packages/29/19/44753eab1fdb50770ac69605527e8859468f3c0fd7dc5a76dd9c4dbd7906/websocket_client-0.56.0-py2.py3-none-any.whl (200kB)
     |################################| 204kB 2.5MB/s
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\08\99\c0\5a5942f5b9567c59c14aac76f95a70bf11dccc71240b91ebf5\intervaltree-3.0.2-cp27-none-any.whl
Collecting colorama
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Processing c:\users\andcho01\appdata\local\pip\cache\wheels\7d\d4\49\fc213a9c521e54184f24f2014f26bfb50ad06d39ab559d4938\pywinusb-0.4.2-cp27-none-any.whl
Collecting idna<2.9,>=2.5
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
[... 31 lines cut ...]
  Downloading https://files.pythonhosted.org/packages/e9/45/9c82d3666af4ef9f221cbb954e1d77ddbb513faf552aea6df5f37f1a4859/pathlib2-2.3.5-py2.py3-none-any.whl
Collecting zipp>=0.5
  Downloading https://files.pythonhosted.org/packages/74/3d/1ee25a26411ba0401b43c6376d2316a71addcc72ef8690b101b4ea56d76a/zipp-0.6.0-py2.py3-none-any.whl
Collecting backports.functools-lru-cache; python_version < "3"
  Downloading https://files.pythonhosted.org/packages/da/d1/080d2bb13773803648281a49e3918f65b31b7beebf009887a529357fd44a/backports.functools_lru_cache-1.6.1-py2.py3-none-any.whl
Collecting scandir; python_version < "3.5"
  Downloading https://files.pythonhosted.org/packages/c6/8c/43cc3799c79c435d1a236783993b2e04a2c750b4f91ef3630ec442490df5/scandir-1.10.0-cp27-cp27m-win32.whl
Collecting more-itertools
  Downloading https://files.pythonhosted.org/packages/2f/9d/dcfe59e213093695f108508af1214cf9cd95cc5489e46877ec5cb56369e5/more_itertools-5.0.0-py2-none-any.whl (52kB)
     |################################| 61kB 1.3MB/s
Building wheels for collected packages: pyOCD, future, pyrsistent
  Building wheel for pyOCD (setup.py) ... done
  Created wheel for pyOCD: filename=pyocd-0.16.0-cp27-none-any.whl size=479529 sha256=f0400b1ae5e81c58861faa109880b2afa51712deb16a008730cd47560c739008
  Stored in directory: C:\Users\andcho01\AppData\Local\pip\Cache\wheels\a3\69\34\2ea4a1fc447bf1f1cceab1f4c0109c7b619be9a8c94a748ff0
  Building wheel for future (setup.py) ... done
  Created wheel for future: filename=future-0.18.2-cp27-none-any.whl size=502599 sha256=e276c44ecef7d11e894f42878c97448f5fd2a67d728eebffd2e1962e02045b15
  Stored in directory: C:\Users\andcho01\AppData\Local\pip\Cache\wheels\8b\99\a0\81daf51dcd359a9377b110a8a886b3895921802d2fc1b2397e
  Building wheel for pyrsistent (setup.py) ... done
  Created wheel for pyrsistent: filename=pyrsistent-0.15.5-cp27-cp27m-win32.whl size=56569 sha256=824d32ccc443a5d20724a9d21da63267ba4e1a1a0adf8ab27fe89fffa213b201
  Stored in directory: C:\Users\andcho01\AppData\Local\pip\Cache\wheels\c4\5a\b2\d22f93864c638398dad50541af906e9838bb22044d2cc3ad3a
Successfully built pyOCD future pyrsistent
[... cut ...] zzy-0.17.0 idna-2.8 importlib-metadata-0.23 intelhex-2.2.1 intervaltree-3.0.2 jsonschema-3.1.1 mbed-cli-1.8.2 mbed-ls-1.6.2 monotonic-1.5 more-itertools-5.0.0 numpy-1.16.5 pathlib2-2.3.5 project-generator-0.9.13 project-generator-definitions-0.2.38 pyOCD-0.16.0 pyYAML-5.1.2 pyelftools-0.25 pyrsistent-0.15.5 pyserial-3.4 pyusb-1.0.2 pywinusb-0.4.2 requests-2.22.0 scandir-1.10.0 six-1.13.0 sortedcontainers-2.1.0 soupsieve-1.9.5 urllib3-1.25.6 websocket-client-0.56.0 xmltodict-0.12.0 zipp-0.6.0
```

0xc0170, 2019-11-07:
I can reproduce this problem. It is just a warning. As pyOCD works with older pyYAML, other package (earlier installed I believe) requests the newest version (I believe this could be progen as we do not have strict version there, I should fix that but that would be now >= 5.1, so this warning would be there anyway).

@flit Would it make sense to update pyOCD to use pyYAML >=5.1 ?

flit, 2019-11-27:
The latest pyocd (0.23.0 as of this writing) requires pyyaml >= 5.1. We just need to update the pyocd used in DAPLink.

andrewc-arm, 2019-11-28:
Hi, @0xc0170 and @flit 
Then it seems we need to change this line to upgrade pyOCD?
https://github.com/ARMmbed/DAPLink/blob/master/requirements.txt#L4

0xc0170, 2019-11-28:
Correct, @andrewc-arm can you please send PR fixing it?

flit, 2019-11-28:
I think it's not quite that simple. The pyOCD API has probably changed some since 0.16.0 (can't remember details, but it's in the [pyOCD release notes](https://github.com/mbedmicro/pyocd/releases)). I was going to attempt to get to it after the gcc port, but if someone else could take care of it that would be fantastic.

andrewc-arm, 2019-11-29:
> @andrewc-arm can you please send PR fixing it?

Sorry. I am new to DAPLink and even if took a hack at it, it would be dangerous. :)
However, I will be mindful of tips and next time when I have to test DAPLink tests I will give it a try and share the opinion here.

0xc0170, 2020-01-08:
> I think it's not quite that simple. The pyOCD API has probably changed some since 0.16.0 (can't remember details, but it's in the pyOCD release notes). I was going to attempt to get to it after the gcc port, but if someone else could take care of it that would be fantastic.

I can have an attempt to at least get initial branch. I'll check it

JanneKiiskila, 2020-01-22:
@0xc0170  - any progress?

0xc0170, 2020-02-03:
> @0xc0170 - any progress?

Not yet, I was OoO most of the last weeks. I'll keep this open to check.

mathias-arm, 2020-12-15:
@flit: I guess since #746 was merged this should be closed.

flit, 2020-12-17:
Yep, thanks.

closed 2020-12-17 by flit