ARMmbed/DAPLink #670: tools\launch_uvision.bat does not work with UV5

andrewc-arm, 2019-11-06:
The batch file is targeted for UV4 instead of UV5.
Someone needs to fix it.

0xc0170, 2019-11-07:
I can't reproduce. Although I am having python 3 (fixing errors in post_build_script), I can get this without errors.

See

```
Build started: Project: k20dx_frdmk64f_if
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'k20dx_frdmk64f_if'
Before Build - User command #1: ..\..\..\tools\pre_build_script.bat
Python 3.7.3
F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if>python ..\..\..\tools\pre_build_script.py 
warning: LF will be replaced by CRLF in tools/flash_algo.py.
The file will have its original line endings in your working directory
#> Pre-build script start
#> Getting git SHA
#> Checking for local changes
#> Keeping git version file since it didn't need to change
#> Pre-build script completed written ..\..\..\source\daplink\version_git.h
linking...
Program Size: Code=44670 RO-data=7446 RW-data=1516 ZI-data=13364  
After Build - User command #1: ..\..\..\tools\post_build_script.bat F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if C:\Keil_v5\ARM
Python 3.7.3
F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if>fromelf --bin F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if.axf -o F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if.bin 
F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if>fromelf --i32 F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if.axf -o F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if.hex 
F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if>python ..\..\..\tools\post_build_script.py F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if.hex F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if_crc 
F:\Code\DAPLink\projectfiles\uvision\k20dx_frdmk64f_if\build\k20dx_frdmk64f_if_crc
board_id None
family_id None
bin_offset None
Start 0x8000, Length 0x17c00, CRC32 0x45442513
".\build\k20dx_frdmk64f_if.axf" - 0 Error(s), 0 Warning(s).
Build Time Elapsed:  00:00:07
```

My setup: using venv, so fresh install for daplink, latest master as it is:

` git logg -1
*   b164e863 - (HEAD -> master, origin/master, origin/HEAD) Merge pull request #661 from ARMmbed/fix_systick_inc (2 weeks ago)`

Python 3.7.3 installed,  `git --version
git version 2.21.0.windows.1` 

uvision 5.28 (I had to migrate to device pack when I opened uvision).  I had to fix few python 2-3 errors, otherwise good

0xc0170, 2019-11-07:
From your second terminal output, what about git - is it present , how did you download daplink?

andrewc-arm, 2019-11-13:
Hi, @0xc0170 
Thanks for your fast feedback. For now, I am able to compile a certain DAPLink repo (zipped shared to me) shared by my partner and failing the one from the latest github repo. This means I have a bypass to lower this priority. So when I have time to follow up on this, I will reopen the case. And it must be my local setup issue. Thank you for your support!

closed 2019-11-13 by andrewc-arm