ARMmbed/DAPLink #683: How to build M484 project?

Kevincoooool, 2019-11-30:
when I  progen generate -t uvision
I got this:
RuntimeError: Target m485sidae is not supported. Please add them to https://github.com/project-generator/project_generator_definitions

Kevincoooool, 2019-12-02:
I changed the target chip to stm32f103xb, then the mdk project can be generated normally,  then downloaded to M484SIDAE, the serial port works normally, and the program can be downloaded to the target chip, but it is very slow and it seems that the internal high-speed PHY is not used

0xc0170, 2019-12-02:
> RuntimeError: Target m485sidae is not supported. Please add them to https://github.com/project-generator/project_generator_definitions

This error messaage comes from project generator, the MCU set for your project is not supported for uvision. You can add it to  https://github.com/project-generator/project_generator_definitions. The project contains more information how to do it.

@ychsu-tf Please review

ychsu-tf, 2019-12-02:
Hi,
Target m485sidae has been added to project_generator_definitions recently.
Please update it to the latest version v0.2.40.

0xc0170, 2019-12-11:
> Please update it to the latest version v0.2.40.

This should be resolved, I'll close the issue.

closed 2019-12-11 by 0xc0170