ARMmbed/DAPLink #684: In application programming aborted due to an out of bounds address

FLATT-Dev, 2019-11-30:
Hello

I have an NXP .iMX RT 1010 eval board and I managed to get it stuck in bootloader mode

When I  plug it into a PC the "Maintenance" drive appears but the COM port does not enumerate.
 I followed instructions in the HELP_FAQ.HTM and tried to flash the board with 3 firmware images suggested in the document. (current and 2 older ones). All fail with the error "In application programming aborted due to an out of bounds address"

The images I tried: 
0241_kl26z_microbit_0x8000.hex ,
0250_kl26z_microbit_0x8000.hex 
0253_kl26z_microbit_0x8000.hex. 

All failed with the error above.
Any ideas how to fix it are much appreciated
Here are the contents of DETAILS.TXT file on the MAINTENANCE drive
Thanks
Andy

# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 00000000161321cf00000000000000000000000097969905
HIC ID: 97969905
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0244
Git SHA: 8ede6eacbce0a3012a899ba13fafe7d400094883
Local Mods: 0
USB Interfaces: MSD
Bootloader CRC: 0xe493996b
Interface CRC: 0x619d86db
Remount count: 3

0xc0170, 2019-12-02:
> Git SHA: 8ede6eacbce0a3012a899ba13fafe7d400094883

I can't find this sha in this repository.

I wonder what bootloader is there, if the offset is not set to expected 0x8000. One way would be to update the bootloader itself (https://github.com/ARMmbed/DAPLink/blob/master/docs/BOOTLOADER_UPDATES.md) .

cc @mmahadevan108 Please review

FLATT-Dev, 2019-12-02:
This is a new NXP i.MX RT 1010 EVK . Perhaps they modified the bootloader
to fit their needs. I have contacted them as well, will see what they have
to say.
Is there a way to update the bootloader without building  from the source
as described in the developer guide?
Thank you

Andy

0xc0170, 2019-12-02:
OK, please update us once you get an answer from NXP (can link the forum here if you asked there).

FLATT-Dev, 2019-12-02:
Sure thing
Andy

FLATT-Dev, 2019-12-04:
Here is the info from the NXP FAE
Worked for me

So far, the MIMXRT1010-EVK using LPC4322JET100 chip as FreeLink main MCU.There with old prototype board using Kinetis chip.
We would recommend you to use the latest version board.
If the board using LPC4322JET100 chip as FreeLink main MCU, the FreeLink is same with  MIMXRT1060-EVK
Customer could check with below link to get related on board debugger firmware:
https://www.nxp.com/design/microcontrollers-developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA#MIMXRT1060-EVK

0xc0170, 2019-12-11:
I assume this is resolved then. I'll close this. Thanks for the info how to fix this!

closed 2019-12-11 by 0xc0170

FLATT-Dev, 2019-12-11:
Yes, resolved
Glad I could help
Regards
Andy