ARMmbed/DAPLink #685: DAPLINK 11U35 for LPC1768/88 over android devices.

fabioriksilva, 2019-12-10:
Hello there!
  I'm here about 3 weeks trying to discover how to solve a problem like above.
  I can't make the daplink work on cell phones or tablet, for example. 
  Could you guys help me with this issue?

  Thank you so much!

0xc0170, 2019-12-11:
@fabioriksilva What is the issue ?

fabioriksilva, 2019-12-11:
Hello there!
   Man, I have made a board of the DAPLink using the processor LPC11U35-501.
   The target is the processor from NXP, model LPC1768/88 (I use both).
   I do work pretty well, but I'm trying to put it on work by android smartphone for example, and the android look on this as an unrecognized device and don't show the mass storage device.
   Based on that, I'm trying to set the DAPlink project to being built on KEIL V5.1, and until now nothing. 
   I just can't understand the path to reach the project like I said!!

   Please, in the name of the universe, could you help that unhappy and poor guy!?

   Thank you so much!

0xc0170, 2019-12-11:
So you plug in board to android phone (what firmware is it running on lpc11u35 ?) ? 

Refer to similar issues already reported: https://github.com/ARMmbed/DAPLink/issues/269 - does your firmware include https://github.com/ARMmbed/DAPLink/pull/401 ?

cederom, 2019-12-11:
@fabioriksilva what version of DAPLink firmware do you use? Android fix was introduced in some recent version only. Make sure you are using latest DAPLink firmware.

Also use firmware with a given target compiled in. Only this will contain UMS endpoint (Mass Storage). Target support contains FlashAlgo for that target and can be used with that particular Target. You cannot use UMS for all targets. In order to program different Targets with one DAPLink firmware you need to use `pyOCD` (+`CMSIS-PACK-MANAGER`).

fabioriksilva, 2019-12-11:
Ah sure, I know it!
  The problem is my absolute lack of knowledge to select a target lpc1768, and compile for daplink using the LPC11u35....
   The firmware that I got here is not done with this issue of error in android!!
   I did get the software package using the process below:
   https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md
   I got success by taking those steps, but by now, I have no idea how to:
   Set the target LPC1768, and interface LPC11u35!!
   The idea is putting my name at some place as well, rsrsrs
   
   PS: Here I have the licensed uV5.1!

  I appreciate it whether you could take me through the process, or, maybe set me to see where I can reach the solution for this issue!

cederom, 2019-12-12:
@fabioriksilva you are confusing several issuer together. Also the subject of the problem does not describe that problem you have just described. Please:

* solve one problem at time.
* use **one** GitHub issue per one problem.
* **You are using really outdated 241 firmware**. Use latest DAPLink firmware (254). **Android UMS fixes were introduced in 252**. Here are official DAPLink Releases: https://github.com/ARMmbed/DAPLink/releases
* If there is no PC1768/1788 Target support yet you will have to add new Target. If official documentation is laconic maybe you will find this more useful: https://github.com/cederom/DAPLink/blob/cederom_doc/doc/DAPLinkHandbook.md to have better inderstanding on the DAPLink architecture.
* Take a look at MBED Ready board based on LPC1768 that may give you some directions https://os.mbed.com/platforms/mbed-LPC1768 on how to work with that Target.

fabioriksilva, 2019-12-12:
@cederom I'm absolutely honored by your attention!
   I see, but the subject is relative to my issues!
   As the topic shows up, I have a DAPlink hardware version 11u35, I use it to write LPC1768/88 and works very fine in windos/linux devices, but, in android relatives not done!
   The questions taken up during the topic was all about the topic!!
   What you are saying is that I do need to create a new topic aslike: How to add the target LPC1768 in the DAPlink 11u35 version?
   To be very honest, I'm more frightened as a dog after a thunder!!
   I have been reading that topic, again and again, and still can't understand the path!!
   Please, help me, I'm really don't know what more to do!!!
   Thank you so much for your help, I really appreciate it!!

cederom, 2019-12-13:
Hey @fabioriksilva no worries we will make it work :-)

You did not provide enough details at first, and then lots of details that could indicate many other issues, thus my several fix propositions :-)

If that 11U35 works fine with that LPC17668/88, then clearly this Target is already supported.

**Update your DAPLink firmware to the latest version 254 and the problem should be gone**. Android "support" was introduced in 252, while you are using 241 (note `Interface Version: 0241` in `DETAILS.TXT`).

fabioriksilva, 2019-12-13:
OH, GOSH, none done!!
  It hasn't the LPC11U35 for LPC1768.. snif snif.
 But, a thing I could see after scratching all paths....
  In the path DAPLink\source\family\nxp\- has the archives for lpc1768....
  It means that whether I could understand the texts, I can add a new board one labeled for example:
  myboard_11u35_lpc1768 and build it.....
  Actually, I got a schematic from net called DAPlink put some things against ESD and protection against different voltages between the HCI and TARGET.
  As you can see annexed.
  Thank you brow, you are awesome!!
  If you have some time, please, help me with that!!
 
  Thank_youE10+15 !!
  
[image]

fabioriksilva, 2019-12-16:
Hey @cederom, do you mind helping me?  
  This weekend, I tried 16 configurations, almost all of them have been built but no one worked after made the boot of the interface with the binary generated.
 I really don't know, what more I can do!! :-:
   :-(

cederom, 2019-12-18:
* Hey @fabioriksilva where did you get that DAPLink firmware configured as LPC11U35 + LPC1768?
* If this is not from a Official Master, then it seems to be some sort of local fork.
* Find the sources for LPC11U35 + LPC1768.
* Use that sources to build LPC11U35 + LPC1768 yourself, whether you have to update by hand to latest release, maybe that fork already is in sync with latest release.

fabioriksilva, 2020-04-27:
Hi there, it's Fabio here again.
   It has been about 4 months, studying, reading, trying all I could.
   I have made changes in the configurations of the SO, in daplink-if chain, in anything that I could.
   

   The message is always the same:
   **error: The interface firmware FAILED to reset/halt the target MCU
   type: target**
 
 Sometimes the message has been changed, but did not write the firmware in the LPC1768 target.

  I have used two examples that are in the projectfiles:

  lpc11u35_c027_if
  lpc11u35_archpro_if

  None of them works!!!

  I could see that the only difference between both is the board file descriptor, anything more is absolutely the same.

  Anyone here that had success in create an interface with LPC11u35 for the target LPC1768?

  Please, help!!!

  Thank you so much!

toyowata, 2020-05-01:
@fabioriksilva 

>I have used two examples that are in the projectfiles:
>
>lpc11u35_c027_if
>lpc11u35_archpro_if
>
>None of them works!!!

I've just used the lpc11u35_archpro_if firmware `0253_lpc11u35_archpro_0x0000.bin` with my Arch Pro board and works fine (flashed mbed-os-example-blinky and ran).

I also build binary from the [latest master branch](https://github.com/ARMmbed/DAPLink/commit/d15cc3d2b39c550842f658593e4b7980f6e06148) source code.  It also works fine.
You can probably check your custom hardware?

fabioriksilva, 2020-05-04:
Man, I'm going to try to be more clear, sorry for some misinformation.
   The hardware is absolutely ok, when I use it with the binary from here, for example:
   **https://github.com/ARMmbed/DAPLink/releases/tag/v0254**

   I did try two firmware from the link above: 
   lpc11u35_c027_if
   lpc11u35_archpro_if
   My DAPLink hardware works wonderfully with the two binaries.

   SO:
   I'm using the licensed keil V5.10.
   I have installed the python 2.7--3.3---3.7---3.8, none made difference.
   I Tried to compile in keil in all manners, none worked in MSD mode, just CMSIS and SERIAL. 

  What is calling me the attention is that, in this struct:
  **C:\DAPLink\source\daplink\settings\settings_rom.c:
    static const cfg_setting_t config_default = {
    .auto_rst = 0,
    .automation_allowed = 0,
    .overflow_detect = 0,
   };**
   It is used to select the default values, it doesn't matter if I change it to 0 or 1, the result is always the same, in the DETAILS.TXT, it always returns 1 on the 3 options....

  SO, when I get the git from here: https://github.com/ARMmbed/DAPLink/, and compile it with My keil 5.1;  
  In Modes: CMSIS, SERIAL, works pretty well.
  In MSD, it doesn't work at all !!! 

 It always returns an error

**error: The interface firmware FAILED to reset/halt the target MCU
type: target**
  
 I'm here wondering,, if the KEIL is messing the code....

 If I'm not asking so much, could you help-me, sending-me the code that you have compiled? 
  PS: I'm using: Windows 7-64  Administrator mode, Python 3.8, Keil V5.1(licensed).

  My Skype direction: fabioriksilva - Brazil.

toyowata, 2020-05-07:
@fabioriksilva 

>If I'm not asking so much, could you help-me, sending-me the code that you have compiled?
>PS: I'm using: Windows 7-64 Administrator mode, Python 3.8, Keil V5.1(licensed).

I think the DAPLink build script doesn't support python 3.8.  I am using python 2.7.15 for DAPLink interface firmware build.  I am using Keil MDK v5.29 on Win 10.

You can see the build step here: 
https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md

The mbedcli_compile script compiles whole project and it will take sometime.  So, I recommend to specify your target project such as:
```
python -u tools/mbedcli_compile.py lpc11u35_archpro_if --clean
```

Then, you can get the binary here:
```
DAPLink\BUILD\LPC11U35_ARCHPRO_IF\ARM-CUSTOM_PROFILE\lpc11u35_archpro_if_crc.bin
```

fabioriksilva, 2020-05-07:
Hey @toyowata !!
  Thank you so much for your wonderful help, I'm very glad for your kind attention!!!
   I was thinking in here how to use the Mbed, because there is no information in the link developers.
   But yesterday, I had an idea, and get more than 12 hours no stopping seeking all I could about the steps of the MSD in the project.
   For my surprise, I have found an Easter egg....
   I really don't know whether is some kind of test as in Indiana Jones, to measure our personal worthy, or it is some kind of "bad-prank" that have put it there just for fun....

  Whatever;;;
  At the archive: C:\DAPLink\source\daplink\settings\settings_rom.c
  Line: 126
  procedure: config_rom_init().
  script: 

> // Read settings from flash if the key is valid
>    if (CFG_KEY == config_rom.key) {
>    uint32_t size = MIN(config_rom.size, sizeof(config_rom));
>    memcpy(&config_rom_copy, (void *)&config_rom, size);
>     }

  It points to : config_rom.size
  Looking about this guy, we get:
  // Configuration ROM
static volatile const cfg_setting_t config_rom __attribute__((section("cfgrom"), zero_init));
  
  It is not funny, that kind of joke...

  I just commented out the code and now the system works like clockwork, from Switzerland, yet.   
    
  I'm very grate for all of you who could help me, but, this is my first time in a community like this one, who takes on a project who is OPEN SOURCE, and could see that no one meal is free...

closed 2020-05-07 by fabioriksilva