ARMmbed/DAPLink #693: Release site: DAPLink firmware for Musca boards

danielRep, 2020-01-15:
Hello guys,

I hope that I'm committing any harm by asking where I can find in the release site the last DAPLink firmware for Arm Musca-B1. I saw commits regarding its port, but I can't find the release anywhere in the website.

Thanks

0xc0170, 2020-01-15:
I recall the support was added after the latest release done. Therefore it has not yet been released. Isn't the firmware shared with the software support for musca b1?

cc @avinashmehtadelhi - please review

avinashmehtadelhi, 2020-01-15:
I remember that there was no release, does a src checkout and build not work out for you?

danielRep, 2020-01-15:
Well I didn't tried that @avinashmehtadelhi . I just simply wanted the firmware to update Musca-B1, and since I saw the release site without it, I asked myself if there were any reason.

danielRep, 2020-01-20:
@avinashmehtadelhi , is it possible for you to leave the firmware for the Musca-B1 attached to this post, since I am having some issues with the build.

danielRep, 2020-01-21:
Sorry, but I closed the issue by miss clicking it. @0xc0170 the software support offers DAPLink v254, which as little bit old afik (at least to work with a pyOCD setup). As I asked @avinashmehtadelhi , is it possible to leave here an attachment with the proper updated firmware.bin?

avinashmehtadelhi, 2020-01-22:
@danielRep unfortunately I do not have the setup with me. I will ask my colleagues if they have it. will let u know by tomorrow.

danielRep, 2020-01-22:
That would be great @avinashmehtadelhi . Thank you!

avinashmehtadelhi, 2020-01-28:
HI
There is no one having the DAPLink binary handy. You can either copy from https://community.arm.com/developer/tools-software/oss-platforms/w/docs/425/musca-b1-firmware-update-qspi-eflash-boot-recovery
or
compile from the src code.

danielRep, 2020-01-28:
Thank you even so @avinashmehtadelhi ! I just go the last version of the firmware in a [issue thread of pyocd ](https://github.com/mbedmicro/pyOCD/issues/808). I will close the issue, even thought that it will be great that the newest DAPLink firmware releases for Musca boards could be on the DAPLink Release website.

closed 2020-01-28 by danielRep