ARMmbed/DAPLink #694: File copy over USB MSC to K66F with DAPLink v254 fails randomly

JanneKiiskila, 2020-01-22:
When compiling the mbed-cloud-client-example, the flashing randomly fails on K66F. The bootloader will spot the firmware checksum not matching, so obviously the flash has not been written correctly. I know the produced image is valid, since if I use pyocd flash - that same image will work correctly.

DAPLink details.txt

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0311000032254e450053800ab52900413f31000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0253
Bootloader Version: 0244
Git SHA: 64359f5c786363065a41ec15c348e3d53568da03
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xbd0abee1
Interface CRC: 0x224a671f
Remount count: 3
URL: http://www.nxp.com/frdm-k66f
```

Error printout from the terminal (minicom):
```
������
Mbed Bootloader
[DBG ] Active firmware integrity check failed
No Update image
[ERR ] Active firmware invalid
Failed to jump to application!
����?��@
```

Image is copied over with `cp ./BUILD/K66F/GCC_ARM/mbed-cloud-client-example-internal.bin /media/jankii01/FRDM-K66FD/ && sync`

Using pyocd flash works,

```
yocd flash ./BUILD/K66F/GCC_ARM/mbed-cloud-client-example-internal.bin
[====================] 100%
0022790:INFO:loader:Erased 200704 bytes (49 sectors), programmed 200704 bytes (49 pages), skipped 245760 bytes (60 pages) at 19.67 kB/s
```

terminal output shows the bootloader starts the application after hash checking.

```
����?��@
Mbed Bootloader
No Update image
[DBG ] Active firmware up-to-date
booting...

Start Device Management Client
Using hardcoded Root of Trust, not suitable for production use.
Starting developer flow
Developer credentials already exist, continuing..
Application ready. Build at: Jan 22 2020 10:46:21
Mbed OS version 99.99.99
mcc_platform_interface_connect()
Connecting with interface: Ethernet
NSAPI_STATUS_CONNECTING
NSAPI_STATUS_GLOBAL_UP
IP: 10.45.3.24
Network initialized, registering...
```

JanneKiiskila, 2020-01-22:
@mmahadevan108  @flit

0xc0170, 2020-02-03:
@mmahadevan108 Would you be able to help?

mmahadevan108, 2020-02-05:
I have not heard of this instability before. 

Could you maybe try with an earlier DAPLink release (maybe v253)