ARMmbed/DAPLink #695: DAPLink fails to build from master

JanneKiiskila, 2020-01-22:
I've tried to follow the [developer documents](https://github.com/ARMmbed/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md) in this repo and do a build, however - it fails.

```
(venv) jankii01@ubuntu:~/mbed/DAPLink$ tools/mbedcli_compile.py K66F --release_folder firmware
import: not authorized `argparse' @ error/constitute.c/WriteImage/1028.
import: not authorized `os' @ error/constitute.c/WriteImage/1028.
import: not authorized `sys' @ error/constitute.c/WriteImage/1028.
import: not authorized `shutil' @ error/constitute.c/WriteImage/1028.
import: not authorized `yaml' @ error/constitute.c/WriteImage/1028.
import: not authorized `mbedcli_tools' @ error/constitute.c/WriteImage/1028.
from: can't read /var/mail/pre_build_script
from: can't read /var/mail/package_release_files
from: can't read /var/mail/post_build_script
tools/mbedcli_compile.py: line 29: syntax error near unexpected token `('
tools/mbedcli_compile.py: line 29: `self_path = os.path.abspath(__file__)'
```
Virtual environment made available, the requirements.txt installed.
Arm compiler 5.06 update 6 is available on the machine.

pip list output
```
(venv) jankii01@ubuntu:~/mbed/DAPLink$ pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package                       Version    
----------------------------- -----------
appdirs                       1.4.3      
attrs                         19.3.0     
backports.functools-lru-cache 1.6.1      
beautifulsoup4                4.8.2      
bs4                           0.0.1      
certifi                       2019.11.28 
chardet                       3.0.4      
colorama                      0.4.3      
configparser                  4.0.2      
contextlib2                   0.6.0.post1
enum34                        1.1.6      
fasteners                     0.15       
functools32                   3.2.3.post2
future                        0.18.2     
fuzzywuzzy                    0.17.0     
idna                          2.8        
importlib-metadata            1.4.0      
intelhex                      2.2.1      
intervaltree                  3.0.2      
Jinja2                        2.10.3     
jsonschema                    3.2.0      
MarkupSafe                    1.1.1      
mbed-cli                      1.8.2      
mbed-ls                       1.6.2      
monotonic                     1.5        
more-itertools                5.0.0      
numpy                         1.16.6     
pathlib2                      2.3.5      
pip                           20.0.1     
prettytable                   0.7.2      
project-generator             0.9.13     
project-generator-definitions 0.2.40     
pyelftools                    0.26       
pyocd                         0.16.0     
pyrsistent                    0.15.7     
pyserial                      3.4        
pyusb                         1.0.2      
PyYAML                        4.2b2      
requests                      2.22.0     
scandir                       1.10.0     
setuptools                    44.0.0     
six                           1.14.0     
sortedcontainers              2.1.0      
soupsieve                     1.9.5      
urllib3                       1.25.8     
websocket-client              0.57.0     
wheel                         0.33.6     
xmltodict                     0.12.0     
zipp                          1.0.0 
```

Using Ubuntu 16.04, `python` version is 2.7.12.

`mbed` version is 1.8.2, which does seem old (but then again, requirements.txt should specify it explicitly).

JanneKiiskila, 2020-01-22:
@flit  - any tips?

0xc0170, 2020-02-03:
@JanneKiiskila I'll try to reproduce this tomorrow locally.

0xc0170, 2020-02-07:
Although I tested on windows, but at least I got it to compile

```
λ python tools/mbedcli_compile.py K66F --release-folder firmware
#> Pre-build script start
#> Getting git SHA
#> Checking for local changes
#> Writing git version file
#> Pre-build script completed written C:\Code\DAPLink\source\daplink\version_git.h
Compiling K66F
[mbed] Auto-installing missing Python modules...
Building project DAPLink (K66F, ARM)
Scan: DAPLink
Compile [  0.3%]: frdmk64f.c
Compile [  0.6%]: ep_agora.c
Compile [  0.9%]: frdmk32w042.c
Compile [  1.3%]: hani_iot.c
Compile [  1.6%]: k26f_bl.c
```

I can try to get virtual box running with ubuntu and test

mbrossard, 2023-02-04:
`mbedcli_compile.py` is now deprecated.

closed 2023-02-04 by mbrossard