ARMmbed/DAPLink #704: RT1050_QSPI DAPLINK Flashing not working.

fredlee12001, 2020-03-10:
We have a RT1050_EVB board with QSPI flash boot enabled. 

DAPLink firmware compiled out within this ARMmbed/DAPLink isn't working.  
I have tried all below three, none of them works for me. 

1. 0254_k20dx_mimxrt1050_evk_qspi_0x8000.bin from release254. 
2. Self compiled k20dx_mimxrt1050_evk_qspi_if_crc_legacy_0x8000_master.bin from latest master branch. 
3. Self compiled k20dx_mimxrt1050_evk_qspi_if_crc_legacy_0x8000_May2019.bin  from some old version. 

There is one working I found from NXP website.  https://www.nxp.com/assets/downloads/data/en/initialization-boot-device-driver-code-generation/k20dx_mimxrt1050_evk_qspi_if_crc_20180419.zip

While I could never found that git-hash source code in this ARMmbed/DAPLink repo. 

@mmahadevan108 ,  seems like some compatibility issue there.   Any ideas how could we fix this here? 

From: |   | Work? | Hash
-- | -- | -- | --
NXP_WEBSite |   | V | 342d091d6d64ad1f8dbb3a5384c08be7fe926f5b
master |   | x | 300c8203a07b3aff916ea86becd16a30f6952e9b
May_2019 |   | x | 483b3b3c19e877b69180249b5b76d339f681624f
V254 |   | x | f499eb6ec4a847a2b78831fe1acc856fd8eb2f28

Below is the test binaries and results.  
[DAPLinks.zip](https://github.com/ARMmbed/DAPLink/files/4311429/DAPLinks.zip)

maclobdell, 2020-03-11:
@mmahadevan108 - can you provide a daplink firmware binary for Quad SPI Flash that has all your latest fixes that we can use for testing?

mmahadevan108, 2020-03-11:
Per the description the issue was seen with the master branch as well. My changes to the Flash Blob was merged into the master branch. 
Are you seeing failures with drag-n-drop. Unfortunately I do not have a board with QSPI flash.

fredlee12001, 2020-03-12:
@mmahadevan108 ,  Yes.  I'm seeing the issue with Drag & Drop.  
Not sure what's the difference there between NXP Website version and this DAPLink repo.

GitYun, 2022-08-03:
@mmahadevan108 hi, I also have the same problem, can't find NXP's Git SHA-1: 342d091d6d64ad1f8dbb3a5384c08be7fe926f5b, the following is the official firmware and boot loader provided by NXP:
[k20dx_mimxrt1050_evk_hyper_if_crc_20180419.zip](https://github.com/ARMmbed/DAPLink/files/9247769/k20dx_mimxrt1050_evk_hyper_if_crc_20180419.zip)
[k20dx_bl_crc.zip](https://github.com/ARMmbed/DAPLink/files/9247773/k20dx_bl_crc.zip)