ARMmbed/DAPLink #707: mbed deploy errors

amitchone, 2020-03-13:
Following on from [this issue](https://github.com/ARMmbed/DAPLink/issues/706):

After bodging pyYAML to version 4.2b1 and running an mbed deploy I'm hit with this stack trace:

```
ERROR: Command errored out with exit status 1:
     command: 'c:\users\adam\desktop\daplink\venv\scripts\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'c:\\users\\adam\\appdata\\local\\temp\\pip-req-build-ratrle\\setup.py'"'"'; __file__='"'"'c:\\users\\adam\\appdata\\local\\temp\\pip-req-build-ratrle\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'c:\users\adam\appdata\local\temp\pip-req-build-ratrle\pip-egg-info'
         cwd: c:\users\adam\appdata\local\temp\pip-req-build-ratrle\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\adam\appdata\local\temp\pip-req-build-ratrle\setup.py", line 30, in <module>
        from pip.req import parse_requirements
    ImportError: No module named req
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Cheers,
Adam

0xc0170, 2020-04-01:
What pip you have installed? I assume not <10. This should be fixed in DAPLink

0xc0170, 2020-05-04:
As no update, I'll close this issue.

closed 2020-05-04 by 0xc0170

eleenest, 2020-05-28:
I am also seeing this (when trying to setup DAPLink on linux

`pip install -r requirements.txt 
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting git+https://github.com/armmbed/manifest-tool.git@v1.4.6 (from -r requirements.txt (line 19))
  Cloning https://github.com/armmbed/manifest-tool.git (to revision v1.4.6) to /tmp/pip-req-build-yVbhx9
  Running command git clone -q https://github.com/armmbed/manifest-tool.git /tmp/pip-req-build-yVbhx9
  Running command git checkout -q 711dea922b36150e5f32c7d168cc20036b72a1af
    ERROR: Command errored out with exit status 1:
     command: /usr/local/google/home/eleenest/git_github/DAPLink/venv_daplink/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-yVbhx9/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-yVbhx9/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-jDGSGX
         cwd: /tmp/pip-req-build-yVbhx9/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-yVbhx9/setup.py", line 30, in <module>
        from pip.req import parse_requirements
    ImportError: No module named req
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
`
I have pip 20.1.1 installed.  

From https://stackoverflow.com/questions/25192794/no-module-named-pip-req
it sounds like pip v10 and above removed the pip.req module?  

There are some suggested workarounds, but it wasn't immediately obvious where to apply the changes (it looks like it is installing "armmbed/manifest-tool.git" to a temp directory and running code in there).  When it finishes that temp directory is removed, though.  
So I'm guessing the changes would need to be made in the code downloaded from armmbed/manifest-tool.git?  

Appreciate any suggestions/workarounds.