ARMmbed/DAPLink #708: Lower bMaxPower in USB descriptor from 500 mA to 100 mA

derhexenmeister, 2020-03-13:
I noticed that CircuitPython boards are visible from the iPad Files app, but the Micro:python boards aren't. AdaFruit lists one possible reason as "...the device must have a USB descriptor that indicates it draws 100mA or less. Many USB descriptors say 500mA even if they don't use that much!"

This capability of seeing the Micro:bit from the Files app could be attractive for educational settings where USB is preferred over Bluetooth.

I believe that this setting is here in the code:
```
DAPLink/source/hic_hal/freescale/kl26z/usb_config.c
68://       <i> when the device is fully operational (bMaxPower)
71:#define USBD_CFGDESC_BMAXPOWER      0xFA
```
I would try changing this to match my Teensy 4.0 CircuitPython:
```
Teensy - 0x32 bMaxPower      (100 mA)
Micro:bit - 0xFA bMaxPower      (500 mA)
```
But I believe that I would need a full license of Keil's ARM MDK due to size restrictions in the lite version. The latest version of DAPlink that Micro:Bit has offered is v0250, so I would probably want to patch that. Is there any way without the non-lite tools? Also is this a bad move for other reasons? I believe that 100 mA is sufficient provided that no additional loads are applied to the Micro:bit board.

Separately I would like to explore https://github.com/bbcmicrobit/micropython/issues/636

derhexenmeister, 2020-03-14:
I was able to hack a Micro:bit build and change the descriptor as indicated above. Although it still worked on Windows 10 it still did not mount on the iPad Files application.

If anyone has any other ideas please let me know, otherwise this issue can be closed since this change is not sufficient.

```
Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x008B	wTotalLength   (139 bytes)
0x05	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)
```

derhexenmeister, 2020-03-15:
I will close this as I was able to try the change myself and it wasn't sufficient to accomplish the goal.

closed 2020-03-15 by derhexenmeister

cederom, 2020-03-15:
This may not be the DAPLink issue but the Apple's USB OTG implementation.

I have tried dozens of different cables and adapters just to access USB Mass Storage or SD Card Reader on iOS devices. Some of them did work long ago and stopped working recently. The reason is "too much USB power consunption". Even with the original Apple's adapter. But those USB devices work out of the box even on Android. So the problem seems Apple/iOS related.

derhexenmeister, 2020-03-15:
It could very well be an iOS or Files issue, but since the exact same setup works with a Teensy 4.0 running CircuitPython and not with the Micro:bit I was thinking that I might be able to make changes to the Micro:bit's DAPlink build to work around it. It could also be a property of the file system itself that the Files app doesn't like. Anyways any suggestions for experiments or for how to debug this are appreciated. I have a J-link and soldering iron so I think I can recover if I load really broken experimental firmware. I don't have a USB protocol analyzer.

cederom, 2020-03-15:
The filesystem on DAPLink is a minimalistic implementation of FAT16. It is not FAT32 and it has no MBR.

I can see two possible problems here:
1. iOS does not handle FAT16 and/or memory drive with no MBR.
2. iOS cannot provide enough power to run DAPLink.

You may quickly verify if 2 is the problemy by cloning the DAPLink drive into a Memory Stick that is known for you to work. You can create a byte-to-byte copy with a Unix `dd` program. If such cloned pendrive works then the problem may be with DAPLink power consumption. If such cloned pendrive does not work you may want to step-by-step modify it to add MBR and maybe convert partition into FAT32. You may want to find an Apple Lightning to USB adaper with external power supply - that would eliminate power consuption as a root cause.. but the only one I found had no external power supply option:

https://www.apple.com/shop/product/MD821AM/A/lightning-to-usb-camera-adapter

There was a filesystem related issue but I have provided a patch that made it work both on Android and FreeBSD. `fsck_msdosfs` did not allow disk to mount because of invalid filesystem flags. Both Android and FreeBSD use BSD implementation of `fsck_msdosfs` and also this one was used on macOS BSD.. so probably iOS will use this one too (not sure about that you need to verify). Here are some notes and hints you may find useful:

https://github.com/microbitedupl/shared/blob/master/fix/2018-microbit-daplink-android-mount_problem/README.md

I am almost sure that this will not be related with filesystem but the USB OTG handling even on the original cable (that I have returned). I have experienced dramatic issues with Memory Sticks on iOS devices while the same devices worked fine on Android and PC. I was unable to read standard Pendrive nor SD card on iPhone/iPad. Still you may want to verify, just to make sure :-)

if you really want to pinpoint the issue I would probably recommend to use Xcode + iOS emulator + iOS debuger in order to see system logs and/or Files application logs / debug information on how it works with DAPLink and what could be the exact problem. I am not sure how and if it is possible to debug system applications on a real iOS hardware, so you may want to try iOS emulator if it allows USB routing to a real hardware..

derhexenmeister, 2020-03-15:
Just a quick note, the Teensy 4.0 running CircuitPython appears as a FAT12 drive when I attach it to a Mac and look with Disk Utility. I will study the links that you provided. It will take me a while since I haven't really delved into file systems before. But at least I can easily compare DAPlink to the Teensy CircuitPython setup.

Teensy has a separate part containing a bootloader https://www.pjrc.com/store/ic_mkl02.html I'm not sure if you can connect a debugger, but at first glance it doesn't look trivial https://forum.pjrc.com/threads/57365-Teensy-4-0-how-to-connect-a-jtag-debugger I just received my Teensy last week because I wanted to try out CircuitPython and it looked like the easiest and most affordable way. Plus the 4.0 board has great performance compared to many similar products.

derhexenmeister, 2020-03-15:
Ok - your tip about the boot signature was very useful. These boards ship with DAPlink v0249 and I've been patching v250 because I misread their site. And neither of those has your fix. But the "latest" on their site https://microbit.org/get-started/user-guide/firmware/ is v0253 which has it. If I update the board to that, then the Files app briefly shows the Micro:bit and I can see the contents but it disappears shortly.

I'm not sure why they don't also have a link to v0254 as shown on https://github.com/ARMmbed/DAPLink/releases Maybe it's worth trying, although the change list doesn't give me great hope.

feria35, 2020-08-04:
Hello, 
@derhexenmeister, I also would like to use microbit on ipad with usb with my students because it would be easier et so much faster than bluetooth to transfert our programs.
I also tried different cables .. desperatly ; and I would like to post my feedback

* My ipad is not "pro", it is an iPad 5th generation 2017 with a standard lightning port. Anyway, I have the same results as you.
* The microbit has the latest firmware v0254.

Here are the 3 cables, I tried :
1. [The official Apple Adapter Photo to usb 3](https://www.amazon.fr/gp/product/B01DGDNL2G/ref=ppx_yo_dt_b_asin_title_o05_s01) : the microbit appears for 2 seconds if the adapter is plugged to the wall with the Apple charger.
2. [A nonamed OTG adapter usb to lightning](https://www.amazon.fr/gp/product/B082YP7V16/ref=ppx_yo_dt_b_asin_title_o05_s00) :  The microbit appears for 2 seconds in the file manager
3. A [START RC CABLE](https://www.amazon.fr/gp/product/B07WX74JJV/ref=ppx_yo_dt_b_asin_title_o06_s00) : lightning - **micro** usb : to connect directly Microbit to ipad : it doesn't work at all. The microbit is not detected at all (even if it is feed by 2 batteries)

I give you screenshots of my experience.
I just wonder if the reason of this failure is the one given by [https://phwallen.github.io/microbit-swift/](https://phwallen.github.io/microbit-swift/) : "IOS devices do not lend themselves to be hard-wired to external hardware via a USB connection. Apple view this type of interface as an external accessory which would have to be licensed under the Apple MFi program."
Can a firmware change this failure ?

[image]
[image]