ARMmbed/DAPLink #711: LPCXpresso55S69 not listed at https://armmbed.github.io/DAPLink/

maclobdell, 2020-05-01:
LPCXpresso55S69 is supported in the last few releases of DAPLink, but when we go to the DAPLink releases page, it does not show up.

toyowata, 2020-05-02:
@maclobdell 

The LPCXpresso55S69 binary is available here:

https://github.com/ARMmbed/DAPLink/blob/gh-pages/firmware/0253_lpc4322_lpc55S69xpresso_0x10000.bin

However, the target is not listed here:
https://github.com/ARMmbed/DAPLink/blob/gh-pages/_data/update.yml

You can probably make pull-request to the gh-pages branch?

maclobdell, 2020-05-04:
@0xc0170 - what is the normal process for this? Is this something that was maintained by the old maintainers? For every new board, do we need to ask partners to update the gh-pages branch?

0xc0170, 2020-05-04:
I dont know the process but looking at the history, we maintain the pages.

@maclobdell would you be able to send a PR adding this target to the yml file?

Thanks @toyowata for the references

flit, 2020-06-09:
The internal release procedure on Confluence and OneNote documents the process, I believe.

We need to do a new release, anyway. It's been a long time. Note that we need to skip version 0255, as this is reserved for micro:bit.

closed 2021-06-09 by mathias-arm