ARMmbed/DAPLink #714: LED activity during WebUSB DAPJs flashing doesn't blink in 0254

microbit-carlos, 2020-06-02:
Raising this issue here as the same version of DAPJs works fine with DAPLink 0253, but doesn't with 0254, but maybe this issue has to be transferred.  

Basically, flashing with DAPJs via WebUSB is meant to blink the DAPLink LED, and this stopped working with DAPLink 0254.

Can be replicated using the DAPJs demo and a micro:bit board:
- https://armmbed.github.io/dapjs/examples/daplink-flash/web.html
- Hex to show heart in the micro:bit LED matrix: [makecodeheart.hex.zip](https://github.com/ARMmbed/DAPLink/files/4717966/makecodeheart.hex.zip)
- DAPLink interface 0253: [0253_kl26z_microbit_0x8000.hex.zip](https://github.com/ARMmbed/DAPLink/files/4717979/0253_kl26z_microbit_0x8000.hex.zip)
- DAPLink interface 0254: [0254_kl26z_microbit_0x8000.hex.zip](https://github.com/ARMmbed/DAPLink/files/4717971/0254_kl26z_microbit_0x8000.hex.zip)

microbit-carlos, 2020-06-02:
@thegecko tagging you to this ticket for visibility, as I am still not quite sure if this is a fix in DAPJs or DAPLink.

microbit-carlos, 2020-08-10:
Quick note to add that serial activity via WebUSB doesn't blink the LED either in 0254.

thegecko, 2020-08-11:
I agree this sounds like a DAPLink issue,  with your repro steps it should really help tie down the change which introduced this regression

thegecko, 2020-09-30:
Also worth mentioning we use the latest DAPjs with older DAPLink firmware and don't see this issue.

microbit-carlos, 2021-03-25:
@flit a partial fix for this issue when WebUSB flashing was included as part of PR #763 in this commit: https://github.com/ARMmbed/DAPLink/pull/763/commits/e4e9b2d8b8fcc711fcfe9c4858701548643c825f

However, the LED still doesn't blink during "partial flashing". In the MakeCode and Python editor we use DAPJs with WebUSB CMSIS-DAP to manually erase/flash specific pages to update the user code only ([JS code here](https://github.com/bbcmicrobit/PythonEditor/blob/v2.1.2/js/partial-flashing.js)). With older versions of DAPLink the LED would flash during partial flashing as well, but it currently doesn't.

flit, 2021-06-28:
The issue is that the call to blink the "HID" LED is missing from the CMSIS-DAP v2 bulk interface handler.

https://github.com/ARMmbed/DAPLink/blob/78c51931f2058e72f5f95ec03c5299a6ff2bdde0/source/usb/bulk/usbd_bulk.c#L78-L84

microbit-carlos, 2024-04-19:
This can be closed as fixed in https://github.com/ARMmbed/DAPLink/pull/1008 and refined in https://github.com/ARMmbed/DAPLink/pull/1019/