ARMmbed/DAPLink #719: details.txt in the latest master doesn't expand all the URL

microbit-carlos, 2020-06-11:
Building from the latest commit in master d15cc3d2b39c550842f658593e4b7980f6e06148 causes issues in the details.txt file and it doesn't expand all the URL parameters.

Building the micro:bit interface and loading it into a micro:bit v1.5 (which has bootloader 0246) produces the following details.txt:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9901360149624e45005b400f000000250000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0254
Bootloader Version: 0246
Git SHA: d15cc3d2b39c550842f658593e4b7980f6e06148
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xa28dc05a
Interface CRC: 0x376dd06a
Remount count: 0
URL: https://microbit.org/device/?id=9901&v=@V
```

You can see the URL is `https://microbit.org/device/?id=9901&v=@V` instead of `URL: https://microbit.org/device/?id=9901&v=0254`

flit, 2020-06-11:
Thanks for the bug report.

mathias-arm, 2021-05-14:
We are hitting the 512 bytes limit. It would be complicated to support longer files.

We could discuss removing a few lines (either unilaterally or conditionally, e.g. when the value is 0), or move some of the content to a distinct file.

mathias-arm, 2021-06-07:
A few points that have been discussed:
- The URL field is in the HTML file and it seems to serve no real purpose here.
- The "Git SHA" does not need to be a full hash.
- The URL at the top is not working so the situation should be fixed or it should be removed.

I was thinking we could remove the URL and compress "Git SHA", "Local mods", and the newer "Compiler" into a single `Build: d15cc3d (modified, armcc)`. That could leave some space to possibly add the name of the firmware, e.g. `kl26z_microbit_if` or part(s) of it.

mathias-arm, 2021-06-09:
While modifying we are modifying are there other things that should be changed? Some HICs now have support for SWO and JTAG support, should we expose that information?

microbit-carlos, 2021-11-03:
There is current the `BOARD_EXTRA_BUFFER` method of increasing the file buffer. Is this still required? Do we need to use it in the micro:bit v1 configuration?

mathias-arm, 2021-11-03:
It's gone with #839.

microbit-carlos, 2022-03-30:
This is resolved in the latest release, thanks everyone! 🎉