ARMmbed/DAPLink #723: Provide DAPLink version via WebUSB

microbit-carlos, 2020-07-09:
Right now the DAPLink version cannot be obtained via WebUSB, having this information would be really useful for the micro:bit online editors and other DAPLink consumers. Quite useful as well if we ever need to request users to update their DAPLink firmware.

flit, 2020-07-09:
In the latest code on master, the DAP_Info command will return the DAPLink version string for the `DAP_ID_FW_VER` property.

flit, 2020-08-02:
Update: I discussed this with the CMSIS team ([#962](https://github.com/ARM-software/CMSIS_5/issues/962)). The `DAP_ID_FW_VER` value is intended to be the CMSIS-DAP protocol version. So changing it to report the DAPLink version is non-compliant. 

Instead, I proposed we define a new `DAP_ID_PRODUCT_FW_VER` (0x07) info value that is dedicated to the product's firmware version. This is implemented in CMSIS PR [#973](https://github.com/ARM-software/CMSIS_5/pull/973).

I've already [implemented](https://github.com/flit/DAPLink/tree/feature/cmsis_dap_version) the change for DAPLink, and will merge when the CMSIS-DAP pull request is merged.

microbit-carlos, 2020-08-03:
Thank you Chris!

@thegecko I guess this is something that could go into DAPJs as well when is all merged.

CC FYI @pelikhan @mmoskal

thegecko, 2020-08-03:
> I guess this is something that could go into DAPJs

Yup, looks like a simple extension

mathias-arm, 2021-04-23:
> I've already [implemented](https://github.com/flit/DAPLink/tree/feature/cmsis_dap_version) the change for DAPLink, and will merge when the CMSIS-DAP pull request is merged.

This was done in #776. Can we close the issue?

thegecko, 2021-04-23:
Unfortunately the chosen address for this data clashes with an existing CMSIS-DAP entry (see https://github.com/ARM-software/CMSIS_5/issues/962#issuecomment-825497410). This should be resolved, ideally.

microbit-carlos, 2021-11-08:
@flit just wanted to confirm if the clash mentioned by Rob is resolved already and the version in `develop` is good to go?

flit, 2022-04-28:
Yes, that's all fixed for some time now.

closed 2022-04-28 by flit