ARMmbed/DAPLink #724: DAPLink binaries for Segger OB ?

blockx, 2020-07-09:
Hi,

I am using a NRF51-DK (see [here ](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF51-DK)) which has a Segger OB J-Link debugger and want to use it with MBED Studio.  Aiui, this means reflashing the debugger on the NRF51-DK with a binary of DAPLink but I do not own a version of the Arm compiler so cannot build it.  I did have a go with the Keil MDK-LITE but there were so many problems with eg python dependencies not installing and invalid versions of pyaml etc etc I just give up.  Like the following :

[ERROR: pyocd 0.16.0 has requirement pyyaml<5.0,>=4.2b1, but you'll have pyyaml 5.3.1 which is incompatible.](url)

Can anyone tell me where I can obtain a ready-built version of DAPLink for the Segger OB debugger in the  NRF51-DK ?

Thanks,

Kevincoooool, 2020-07-10:
Hello, you better backup the JLINKOB firmware of the current board first, and then flash the DAPLink firmware. I have the compiled SAM3U DAPLink firmware, but I want to try the Jlink OB SAM3U firmware. Can you read it?

blockx, 2020-07-10:
Hi Kevin,

The J-LINK software for the NRF52-DK works and is available as a download from [here](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52-DK/Download#infotabs).  I then updated to the latest version by running J-Link Commander which is a free download from [here](https://www.segger.com/downloads/jlink/).

In case anyone is interested, I updated to the DAPLink software by entering NRF51-DK into the "Name of board" box [here](https://armmbed.github.io/DAPLink/).

Thanks for the replies,

closed 2020-07-10 by blockx