ARMmbed/DAPLink #726: write fails on Linux and FreeBSD

trombik, 2020-07-22:
#309 states that:
> Closing issue. DAPLink is OS-agnostic firmware. If there are any bugs that prevent it from working with FreeBSD or any other OS, then we will track and fix the bugs as separate issues.

Here is the issue. I cannot successfully flash micro:bit on FreeBSD and Ubuntu. `mount -o sync` does not help.

#### the micro:bit

```
> cat /mnt/DETAILS.TXT 
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9901000052324e45003c8018000000720000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0253
Bootloader Version: 0243
Git SHA: 64359f5c786363065a41ec15c348e3d53568da03
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0x53375800
Remount count: 5
URL: https://microbit.org/device/?id=9901&v=0253
```

#### FreeBSD

The `dmesg` on FreeBSD.

```
ugen0.5: <ARM BBC micro:bit CMSIS-DAP> at usbus0
umass0 on uhub0
umass0: <USBMSC> on usbus0
umass0:  SCSI over Bulk-Only; quirks = 0x8100
umass0:1:0: Attached to scbus1
da0 at umass-sim0 bus 0 scbus1 target 0 lun 0
da0: <MBED VFS 0.1> Removable Direct Access SCSI-2 device
da0: Serial Number 9901000052324e45003c8018000000720000000097969901
da0: 1.000MB/s transfers
da0: 64MB (131200 512 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
umodem0 on uhub0
umodem0: <mbed Serial Port> on usbus0
umodem0: data interface 2, has CM over data, has break
ng_ubt: HCI command 0xfc05 timed out
uhid0 on uhub0
uhid0: <CMSIS-DAP> on usbus0
ng_ubt: HCI command 0xfc05 timed out
```

```
# mount_msdosfs /dev/da0 /mnt
> ls -alh /mnt
total 16
drwxr-xr-x   1 root  wheel   1.0K Jan  1  1980 ./
drwxr-xr-x  21 root  wheel   1.0K Jul 21 09:18 ../
-r-xr-xr-x   1 root  wheel   495B Mar 23  2016 DETAILS.TXT*
-r-xr-xr-x   1 root  wheel   284B Mar 23  2016 MICROBIT.HTM*
# cp ~/Downloads/OutOfBoxExperience-v2.hex /mnt
```

In `/var/log/messages`:

```
Jul 22 17:54:02 x270 kernel: g_vfs_done():da0[WRITE(offset=512, length=4096)]error = 6
Jul 22 17:54:02 x270 kernel: g_vfs_done():da0[WRITE(offset=33792, length=4096)]error = 6
Jul 22 17:54:02 x270 kernel: g_vfs_done():da0[WRITE(offset=67072, length=1024)]error = 6
Jul 22 17:54:02 x270 kernel: g_vfs_done():da0[WRITE(offset=1137152, length=12288)]error = 6
Jul 22 17:54:02 x270 kernel: g_vfs_done():da0[READ(offset=512, length=4096)]error = 6
```

from the man page:

```
     6 ENXIO Device not configured. Input or output on a special file referred
             to a device that did not exist, or made a request beyond the
             limits of the device.  This error may also occur when, for
             example, a tape drive is not online or no disk pack is loaded on
             a drive.
```

```
> uname -msr
FreeBSD 13.0-CURRENT amd64
```

#### Linux 

```
Jul 22 07:27:01 localhost kernel: [   52.956967] usb 3-1: new full-speed USB device number 2 using xhci-hcd
Jul 22 07:27:02 localhost kernel: [   53.106952] usb 3-1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice=10.00
Jul 22 07:27:02 localhost kernel: [   53.106964] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul 22 07:27:02 localhost kernel: [   53.106972] usb 3-1: Product: "BBC micro:bit CMSIS-DAP"
Jul 22 07:27:02 localhost kernel: [   53.106978] usb 3-1: Manufacturer: ARM
Jul 22 07:27:02 localhost kernel: [   53.106986] usb 3-1: SerialNumber: 9901000052324e45003c8018000000720000000097969901
Jul 22 07:27:02 localhost kernel: [   53.234467] usb-storage 3-1:1.0: USB Mass Storage device detected
Jul 22 07:27:02 localhost kernel: [   53.235039] scsi host0: usb-storage 3-1:1.0
Jul 22 07:27:02 localhost kernel: [   53.243411] hid-generic 0003:0D28:0204.0001: hiddev96,hidraw0: USB HID v1.00 Device [ARM "BBC micro:bit CMSIS-DAP"] on usb-xhci-hcd.1.auto-1/input3
Jul 22 07:27:02 localhost kernel: [   53.291396] cdc_acm 3-1:1.1: ttyACM0: USB ACM device
Jul 22 07:27:02 localhost kernel: [   53.294157] usbcore: registered new interface driver cdc_acm
Jul 22 07:27:02 localhost kernel: [   53.294166] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
Jul 22 07:27:02 localhost kernel: [   53.295470] usbcore: registered new interface driver uas
Jul 22 07:27:03 localhost kernel: [   54.245484] scsi 0:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
Jul 22 07:27:03 localhost kernel: [   54.245877] sd 0:0:0:0: Attached scsi generic sg0 type 0
Jul 22 07:27:03 localhost kernel: [   54.246352] sd 0:0:0:0: [sda] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
Jul 22 07:27:03 localhost kernel: [   54.246919] sd 0:0:0:0: [sda] Write Protect is off
Jul 22 07:27:03 localhost kernel: [   54.246926] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
Jul 22 07:27:03 localhost kernel: [   54.247918] sd 0:0:0:0: [sda] No Caching mode page found
Jul 22 07:27:03 localhost kernel: [   54.253295] sd 0:0:0:0: [sda] Assuming drive cache: write through
Jul 22 07:27:03 localhost kernel: [   54.338963]  sda:
Jul 22 07:27:03 localhost kernel: [   54.340945] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

```
# mount /dev/sda /mnt
# cp /home/trombik/OutOfBoxExperience-v2.hex /mnt/
```

```
Jul 22 07:32:03 localhost kernel: [  354.666961] VFS: busy inodes on changed media or resized disk sda
Jul 22 07:32:03 localhost kernel: [  354.679557] sd 0:0:0:0: [sda] tag#0 access beyond end of device
Jul 22 07:32:03 localhost kernel: [  354.685570] blk_update_request: I/O error, dev sda, sector 131072 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Jul 22 07:32:03 localhost kernel: [  354.696614] sd 0:0:0:0: [sda] tag#0 access beyond end of device
Jul 22 07:32:03 localhost kernel: [  354.702675] blk_update_request: I/O error, dev sda, sector 131072 op 0x0:(READ) flags 0x0 phys_seg 8 prio class 0
Jul 22 07:32:03 localhost kernel: [  354.713095] Buffer I/O error on dev sda, logical block 131072, async page read
Jul 22 07:32:03 localhost kernel: [  354.720348] Buffer I/O error on dev sda, logical block 131073, async page read
Jul 22 07:32:03 localhost kernel: [  354.727596] Buffer I/O error on dev sda, logical block 131074, async page read
Jul 22 07:32:03 localhost kernel: [  354.734844] Buffer I/O error on dev sda, logical block 131075, async page read
Jul 22 07:32:03 localhost kernel: [  354.742091] Buffer I/O error on dev sda, logical block 131076, async page read
Jul 22 07:32:03 localhost kernel: [  354.749339] Buffer I/O error on dev sda, logical block 131077, async page read
Jul 22 07:32:03 localhost kernel: [  354.756587] Buffer I/O error on dev sda, logical block 131078, async page read
Jul 22 07:32:03 localhost kernel: [  354.763834] Buffer I/O error on dev sda, logical block 131079, async page read
Jul 22 07:32:05 localhost kernel: [  356.805977] VFS: busy inodes on changed media or resized disk sda
Jul 22 07:32:05 localhost kernel: [  356.806341] sd 0:0:0:0: [sda] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
Jul 22 07:32:34 localhost kernel: [  385.737577] VFS: busy inodes on changed media or resized disk sda
Jul 22 07:32:34 localhost kernel: [  385.748587] sd 0:0:0:0: [sda] tag#0 access beyond end of device
Jul 22 07:32:34 localhost kernel: [  385.754568] blk_update_request: I/O error, dev sda, sector 131072 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Jul 22 07:32:34 localhost kernel: [  385.765346] sd 0:0:0:0: [sda] tag#0 access beyond end of device
Jul 22 07:32:34 localhost kernel: [  385.771326] blk_update_request: I/O error, dev sda, sector 131072 op 0x0:(READ) flags 0x0 phys_seg 8 prio class 0
Jul 22 07:32:34 localhost kernel: [  385.781694] Buffer I/O error on dev sda, logical block 131072, async page read
Jul 22 07:32:34 localhost kernel: [  385.788951] Buffer I/O error on dev sda, logical block 131073, async page read
Jul 22 07:32:34 localhost kernel: [  385.796199] Buffer I/O error on dev sda, logical block 131074, async page read
Jul 22 07:32:34 localhost kernel: [  385.803449] Buffer I/O error on dev sda, logical block 131075, async page read
Jul 22 07:32:34 localhost kernel: [  385.810696] Buffer I/O error on dev sda, logical block 131076, async page read
Jul 22 07:32:34 localhost kernel: [  385.817946] Buffer I/O error on dev sda, logical block 131077, async page read
Jul 22 07:32:34 localhost kernel: [  385.825193] Buffer I/O error on dev sda, logical block 131078, async page read
Jul 22 07:32:34 localhost kernel: [  385.832442] Buffer I/O error on dev sda, logical block 131079, async page read
Jul 22 07:32:36 localhost kernel: [  387.846796] VFS: busy inodes on changed media or resized disk sda
Jul 22 07:32:36 localhost kernel: [  387.847352] sd 0:0:0:0: [sda] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)

```

```
> cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04 LTS"

> uname -msr 
Linux 5.4.45-sunxi64 aarch64
```

cederom, 2020-07-22:
Hello @trombik :-) I am using DAPLink and pyOCD on FreeBSD with no problem. Can you make sure that your firmware file is not too big? DAPLink has implemented virtual file system with only one 512k sector if I remember correctly.
* Can you try with a small firmware file size?
* Can you try with a 12.1 release or stable just to make sure this is not the OS problem?

trombik, 2020-07-23:
> Can you make sure that your firmware file is not too big? DAPLink mass storage endpoint has implemented virtual file system with only one 512k sector if I remember correctly (I may be wrong here).
> 
> * Can you try with a small firmware file size?

Flashing Heart from makecode is probably the simplest, and is 623 KB. I don't have any other file to test.

```
-rw-r--r--   1 trombik  wheel   623K Jul 23 21:30 microbit-Flashing-Heart.hex
-rw-r--r--   1 trombik  wheel   521K Jul 22 20:40 OutOfBoxExperience-v2.hex
```

> * Can you try with a 12.1 release or stable just to make sure this is not the OS problem?

All my machines runs CURRENT. I need to find a box to install first.

> * Make sure to umount the target right after flashing as it will detach itself you need to umount and mount it again (you may like the `sysctl vfs.usermount=1` that will allow your user to mount the drives).

I did.

> * Please check the drive after flashing for any additional files that would indicate an error.

`FAIL.txt` only.

```
error: The transfer timed out.
type: transient, user
```

> * Are you sure your DAPLink has correct firmware flashed and matches your HIC and Target?

The firmware is factory default. 

> You may also want to try out the https://github.com/mbedmicro/pyOCD to flash your target using the `CMSIS-DAP` endpoint (different than `UMS`) of the DAPLink just to make sure your setup is correct :-)

when I have time to do so, I will. did you successfully flash a firmware on micro:bit? I guess you are not using micro:bit.

cederom, 2020-07-23:
Some more ideas:
* You may want to generate (or convert to) a BIN file that will be smaller in size than HEX (below 512K). The most probable issue here is that firmware file size is over the capacity of the handled sector supported by DAPLink UMS module. Using smaller firmware file will confirm or deny that cause.
* You may want to use anything from existing examples code to try out different firmware file below 512K (use https://makecode.microbit.org/). This will not only confirm/deny the file size issue, but also confirm/deny another cause that may be related to invalid firmware image flashed to that target.
* If you are short on time then use pyOCD for flashing, it will take you only several minutes to setup dedicated virtualenv and you will gain very powerful flashing and debug utility for virtually every ARM-Cortex core that is out there with that tiny DAPLink you will find anywhere :-) 
* You can update the DAPLink interface firmware to v254 (https://github.com/ARMmbed/DAPLink/releases) and see if problem persists.

trombik, 2020-07-24:
> Some more ideas:
>
> * You may want to generate (or convert to) a BIN file that will be smaller in size than HEX (below 512K). The most probable issue here is that firmware file size is over the [...]

I don't have any. If you do, I will appriciate.

> * You may want to use anything from existing examples code to try out different firmware file below 512K (use https://makecode.microbit.org/). This will not only confirm/deny the file size issue, [...]

As I wrote, the code is from the makecode. The flashing heart was chosen because it is simple. I don't know any other smaller examples.

> * If you are short on time then use pyOCD for flashing, it will take you only several minutes to setup dedicated virtualenv and you will gain very powerful flashing and debug utility for virtually every ARM-Cortex core that is out there with that tiny DAPLink you will find anywhere :-)

maybe in the future.

> * You can update the DAPLink interface firmware to v254 (https://github.com/ARMmbed/DAPLink/releases) and see if problem persists.

if you cannot write a application code, how can I update the firmware?